ArduPilot/ardupilot #15691: [Plane] Throttle nudge pulsing motor in CRUISE/RTL

giacomo892, 2020-10-30:
Automatic throttle works fine as soon throttle (rc3) is below 50%. As soon I ask for a nudge (rising throttle stick above 50%)  the throttle starts pulsing very fast limited only by TRIM_THROTTLE and THR_MAX.

**Version**
Master 379506b718

Throttle nudge was working fine in 29e6f058a9

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Flying Wing

**Hardware type**
Omnibus F4 PRO

**Logs**

https://mega.nz/file/x0w0UD7S#JXSTFEPbGFfsQNqcqTgAokRYk6zG4r45YBRZcAox2aM

Unfortunately the motor output log seems like filtered and it doesn't show the issue very well, but the interested areas are when I rise RC3 and I am in CRUISE/RTL, but I guess TECS log should have more info which I am not able to understand.
I have also a DVR showing the issue if needed and I can share it with devs.

[image]

giacomo892, 2020-10-31:
Good news, I can reproduce the issue. Did another flight and it showed a similar behavior.

IamPete1, 2020-10-31:
[image]

The throttle is oscillating to full when TECS is trying to accelerate, I cant immediately see anything that would cause this. There have been no major changes to TECS between the versions you have given. Some of the TECS param are not default but I don't see that the changed values should cause this.

giacomo892, 2020-10-31:
@IamPete1 Thanks for taking a look, I will try to revert to that revision ASAP and fly again

giacomo892, 2020-11-01:
I've compiled a firmware based on the old revision (29e6f05) and it is working as expected.
So there is definitely a bug somewhere.

Log: https://mega.nz/file/d4hCha7K#faHwV76lnSJCm0xi1cQ4IT3XHwl1ismMKGb1VAZpnEY

giacomo892, 2020-11-03:
I am still in the progress of bisecting in order to find the root cause here is where I am now
Working revision: b66eaa0d53675f6aef9b54880e5a5f961f915faf (7 aug 2020)
Bad revision:c265a383448500a0008152e6b550f2c213b7c606 (26 aug 2020)

Can you spot something before I do the next step?

IamPete1, 2020-11-03:
Can you reproduce in SITL?

giacomo892, 2020-11-03:
> Can you reproduce in SITL?

I tried but no, might be a fault from my side too

Narrowed the range after another test flight:
Bad revision: c722367c6c48a01260e7bf6af4edc0e513af5132 (12 aug)
Working revision: b66eaa0d53675f6aef9b54880e5a5f961f915faf (7 aug)

giacomo892, 2020-11-06:
https://github.com/ArduPilot/ardupilot/commit/62354527d46066eeea9ac72ce36024391c8eb8d6

This might be the culprit, I think we should move 

```
        airspeed_nudge_cm = 0;
        throttle_nudge = 0;
```

inside the control_failsafe() and apply it if only rc_failsafe_active is true

Will test it and report if it is the offending commit.

IamPete1, 2020-11-06:
Glad you found it, I think putting it in here would probably be a fair change, https://github.com/ArduPilot/ardupilot/blob/ed7580e83c2ad50ca0b2727ffb6a2beee1a0237a/ArduPlane/radio.cpp#L260

Interesting that your receiver is hitting that at all, might be worth investigating that further.

PR #15743 (2020-11-09): ArduPlane: Fix throttle nudge reset due to wrong radio timing
Fixes #15691 

Is this ok for you @IamPete1 ?
closed 2020-11-09 by tridge