ArduPilot/ardupilot #15693: SYSID_MYGCS is not fully explained.

muramura, 2020-10-31:
## Bug report

Recently, the definition of the config parameter "SYSID_MYGCS" has been changed.
This change removes the description of the already major Mission Planner and APM PLANNER2 numbers in the GCS number.
I could not find an entry in APM PLANNER2 to set the GCS number.
I think it would be better to describe the major GCSs in the description so that they are not assigned the same number.

PR:
https://github.com/ArduPilot/ardupilot/pull/15690

**Version**
ArduCopter V4.1.0-dev (fee14922)

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
QUAD

**Hardware type**
SITL
**Logs**

WickedShell, 2020-11-02:
This was intentional in #14259, and for the reason's I closed your PR I don't think we should really revisit it, but we can have the discussion.

peterbarker, 2020-11-02:
The descriptions were not common across vehicles.

The descriptions are misleading in that there's nothing magical about the IDs - they're not *allocated* to a GCS.

Nominating specific values for a GCS in our parameters misleads in that people may not realise that connecting two MissionPlanner instances is as easy as changing the ID, or that we might treat these identifiers specially.

I contend the issue topic here is incorrect, at least as far as adding any special values in. 

(Sadly, APM Planner2 doesn't seem to have any UI for changing its systemID).

IamPete1, 2021-08-16:
Since we close the PR I think we should also close this issue.

closed 2021-08-16 by IamPete1