ArduPilot/ardupilot #15696: SITL Quad Copter, 4.1.0-dev (379506b7): Heavy oscillations on pitch and roll

hhackbarth, 2020-10-31:
## Bug report

**Issue details**

When using SITL with current 4.1.0-dev (379506b7) with default settings for "quad", the copter heavily oscillates on pitch and roll. You can see that already when simply taking of to - lets say 10m - and letting it hover there. The HUD horizon shakes all the time. In the logs you see heavy oscillations on pitch and roll and as a result also permanently alternating RCOU values on the motor channels.

Pitch & Roll:
https://live.staticflickr.com/65535/50550714447_2948ecab6b_h.jpg

RCOU:
https://live.staticflickr.com/65535/50549852423_c25cecbb2f_h.jpg

**Version**
4.1.0-dev (379506b7)

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quad

**Hardware type**
SITL

**Logs**
https://gist.githubusercontent.com/hhackbarth/0fb3d9cb945d168823aa3f36f8439568/raw/11c46af441973e30d24077016e30c1dcf698f81e/2020-10-31%252013-27-49.log

IamPete1, 2020-10-31:
Just tested here and its working fine for me, I would guess it is something on your end. Make sure your git its upto date and your params are defaults.

hhackbarth, 2020-10-31:
Thanks for testing on your side.
Reason seems to be, that I changed FRAME_TYPE from 0 to 1 (from "Plus" to "X"). This was not a problem until now.
Don't know, why it was flying fine before and not yet anymore and if this is a bug. Changing back to 0 solved the issue.
If you (developers) think, that the new behaviour is on purpose, than this issue can be closed.

IamPete1, 2020-10-31:
Its due to this I would guess, https://github.com/ArduPilot/ardupilot/pull/15613

I think that counts as on purpose.

closed 2020-10-31 by IamPete1

rmackay9, 2020-11-02:
@hhackbarth,

I've tested the X frame and it seems to work OK.  I suspect the issue is that in your setup the FRAME_TYPE parameter has been set to "1" but the SITL model is still using a plus frame.  Be sure to start SITL with "-f X" appended.

>> ../Tools/autotest.sim_vehicle.py -f X --map --console