ArduPilot/ardupilot #15699: SITL: gimbal rotate resets copter's yaw to 0

hhackbarth, 2020-10-31:
## Bug report

**Issue details**
When a MAVLink command is issued to rotate the gimbal, the copter's yaw is reset to 0 degree.

I am using Dronekit 2.9.2 which sends MAVLink commands through pymavlink. The rotate(pitch, roll, yaw) function for the gimbal contains the following code sequence:
```
        msg = self._vehicle.message_factory.mount_configure_encode(
            0, 1,    # target system, target component
            mavutil.mavlink.MAV_MOUNT_MODE_MAVLINK_TARGETING,  #mount_mode
            1,  # stabilize roll
            1,  # stabilize pitch
            1,  # stabilize yaw
        )
        self._vehicle.send_mavlink(msg)
        msg = self._vehicle.message_factory.mount_control_encode(
            0, 1,    # target system, target component
            pitch * 100,  # pitch is in centidegrees
            roll * 100,  # roll
            yaw * 100,  # yaw is in centidegrees
            0  # save position
        )
        self._vehicle.send_mavlink(msg)

```
This seems to happen only in SITL, not in the real world. I tested with MNT_TYPE == 4

**Version**
ArduCopter 4.1.0-dev, Dronekit 2.9.2

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
SITL in Mission Planner

**Logs**

muramura, 2020-11-26:
The message is broadcast to the FC for processing.
MNT_TYPE is 4.
You need to set the system ID and component ID of the gimbal.
FC will route messages that are not addressed to it.
GCS<-->FC<-->GYMBAL

muramura, 2020-11-26:
STORM32 does not seem to have PAN control.

rrr6399, 2020-11-30:
> STORM32 does not seem to have PAN control.

It just has a hard-coded value of false for has_pan() control. If you hard code it to true, the gimbal works as expected and the copter does not yaw.

amilcarlucas, 2021-01-26:
Is this a bug? Can it be fixed with a PR?

IamPete1, 2021-08-08:
This seems to be expected behavior for the 2 axis gimbal in SITL.

rmackay9, 2022-09-05:
I've reproduced this issue and I have a fix in PR https://github.com/ArduPilot/ardupilot/pull/21639

rmackay9, 2022-09-06:
fixed in master and the fix will be included in Copter-4.3

closed 2022-09-06 by rmackay9