ArduPilot/ardupilot #15700: New Motor Geometry Mixer

PASC-ERT, 2020-11-02:
Feature Request:

We're working on a new copter design. It is a Tri-Copter (Y3 and Y6). There is already such a motor mix geometry. Ours is different in such a way that instead of the current geometry (2 motors in the front and one in the back) ours will have one motor in the front and 2 in the back.

Would it be possible to add this new geometry motor mix to the Y-Configuration pages.

Basically and ideally would be to take the current Y-Configuration with all the futures and flip it around 180 degrees on the X-Axis.

**Platform**
[ ] All
[ ] AntennaTracker
[X] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

IamPete1, 2020-11-02:
https://discuss.ardupilot.org/t/reverse-tricopter-vtol-plane/39829

For a tradition Tri-Copter with a yaw servo, set FRAME_TYPE to 6.

I think this is what you want, or do you mean a coaxial Y6 with no servo?

auturgy, 2020-11-02:
> For a traditional Tri-Copter with a yaw servo, set FRAME_TYPE to 6.

I think he means the y is flipped (V at back, I in front). The only real effect this has (on tri or y6) is inverting pitch response

PASC-ERT, 2020-11-02:
I mean that when you look at a frame from above, the current config has 2 motors in the front and one in the back. The new proposed geometry is having 2 motors in the front and one in the back. Yaw servo on the front motor is acceptable. See attached diagram (not to scale)[image]

IamPete1, 2020-11-02:
Yes, that is already supported, setup as with normal tricopter and set FRAME_TYPE to 6.

closed 2020-11-02 by IamPete1