ArduPilot/ardupilot #15703: QuadPlane/Tailsitter: DesYaw is not set to Yaw when switched to QStabilize

kcwian, 2020-11-02:
## Bug report

**Issue details**
Before takeoff, when I switch mode to QStabilize in my flying wing tailsitter, the DesYaw is not set to Yaw. When armed, the yaw controller integrates the error and causes the significant deflection of servos, especially when Q_TAILSIT_THSCMX > 1. 
I think it is a bug, as it makes it very hard to takeoff, because the same servos control pitch and yaw axis. As a workaround I found that I can switch to QHover first, then to QStabilize and DesYaw will be set to Yaw. A solution is also to rotate the wing in yaw axis to reduce servos deflection, but sometimes I forget to do so. Solving this issue can probably make a life easier to someone new to tailsitters..

**Version**
ArduPlane 4.0.6

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
VTOL - Tailsitter - flying wing

**Hardware type**
Pixhawk

**Logs**
[image]

IamPete1, 2020-11-03:
This is because we don't relax the controllers to allow a belly sitter to take off. It is expected behavior, however we probably could handle it better. This is a PR that will fix it for yaw and roll: https://github.com/ArduPilot/ardupilot/pull/15195

IamPete1, 2021-07-26:
We now relax correctly for normal tailsitters.

closed 2021-07-26 by IamPete1