ArduPilot/ardupilot #15711: COMPASS2_ORIENT apparently ignored

arikrupnik, 2020-11-02:
## Bug report

**Issue details**

_Please describe the problem_

I have a an external GPS/compass unit that is mounted at 90 degree CCW offset from its nominal orientation. Changing the value of `COMPASS2_ORIENT` parameter seems to have no visible effect on compass readings. They are 90 degrees off. The FC (PH4m) has an onboard compass (in default orientation). With both compasses enabled, I get correct headings. But if I disable the first (built-in) compass, I get readings that are off by 90 degrees.

There's some discussion of the issue on the forum: https://discuss.ardupilot.org/t/compass2-orient-6-yaw270/63576

Photo of the installation:
[image]

**Version**
_What version was the issue encountered with_
Plane 4.0.6 

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Eflite Scimitar; flying wing

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_
Holybro PixHawk 4 Mini; mRo compass/GPS unit

**Logs**
_Please provide a link to any relevant logs that show the issue_

arikrupnik, 2020-11-02:
I should add that I tried changing `COMPASS2_ORIENT` to every single value and power-cycle the AP after each value write. I get the same readings regardless of the value.

rmackay9, 2020-11-02:
It's likely that the COMPASS2_EXTERNAL = 0.  The orientation aren't applied for internal compasses.  It would be good to confirm 100% that the 2nd compass is actually the external compass that you think it is.

rmackay9, 2020-11-02:
I'm going to close this because it's really a support issue and should be raised in the forums first, confirmed to be a bug or enhancement and then raised here.  Thanks for the report though.

closed 2020-11-02 by rmackay9

arikrupnik, 2020-11-03:
> It's likely that the COMPASS2_EXTERNAL = 0. The orientation isn't applied to internal compasses. It would be good to confirm 100% that the 2nd compass is actually the external compass that you think it is.

Thank you for for following up, @rmackay9 . If this is a support issue, perhaps I can impose on you to weigh in on my forum thread.

COMPASS2 is set as external. Here's a screenshot from the thread:

[image]