ArduPilot/ardupilot #15716: Quadplane doing Early VTOL transition in Guided Mode SITL

SaiHaneeshAllu, 2020-11-03:
Hi. I'm trying to navigate the Quadplane ( ArduPlane V4.1.0dev (f18585c5 ) in Mission Planner SITL in Guided Mode.  The following parameters are set for it 

Q_ENABLE = 1
Q_GUIDED_MODE = 1
WP_MAX_RADIUS = 0
WP_RADIUS = 90
WP_LOITER_RAD = 80 
Q_RTL_MODE = 1

As per the documentation in [quadplane](https://ardupilot.org/plane/docs/quadplane-auto-mode.html#guided-mode)  "The approach to the guided waypoint will be done as a fixed wing aircraft. The transition to VTOL flight will begin at the WP_LOITER_RAD radius in meters. This should be set appropriately for your aircraft. A value of 80 meters is good for a wide range of QuadPlanes."   But the VTOL transition is happening when the vehicle is at around 150-160 meter.  I even tried changing the parameters as below:

WP_RADIUS = 60
WP_LOITER_RAD = 60 . But it still the transition is happening at around 130-140 meter. Is there any parameter that i missed for this transition to take place as expected or is this a real issue ? 

Attached are the pictures of flight and transition well ahead of the waypoint radius in guided mode

[image]
[image]

IamPete1, 2020-11-03:
I would guess its fly such that the transition finishes at the radius, have you tried increasing `Q_TRANS_DECEL`?

SaiHaneeshAllu, 2020-11-03:
Hi @IamPete1 , i just increased `Q_TRANS_DECEL` to 4 from the default 2. But still the transition is happening about the same distance ( 150-160m).

robustini, 2020-11-03:
So I'm not the only one, see https://github.com/ArduPilot/ardupilot/issues/15666

robustini, 2020-11-03:
Try to increase Q_TRANS_DECEL to 5.

SaiHaneeshAllu, 2020-11-03:
Hi @robustini , I increased `Q_TRANS_DECEL` to the maximum value 5. But after doing simulations, i found that this increment has improved performing transition at the distance specified by  `WP_LOITER_RAD`  while in RTL only but not in GUIDED mode. In guided mode it's happening around the same 150 meter mark despite the increment.  Would you please try a simulation in guided mode and confirm if its performing transition at near specified radius. That would be of great help to me ! Thanks

SaiHaneeshAllu, 2020-11-03:
I just tried setting the `WP_LOITER_RAD` and `WP_RADIUS` to 20m (although not desirable) each. Now it is performing the transition at 90m mark to the waypoint.

SaiHaneeshAllu, 2020-11-04:
> Hi @robustini , I increased `Q_TRANS_DECEL` to the maximum value 5. But after doing simulations, i found that this increment has improved performing transition at the distance specified by [...]

[image]

[image]