ArduPilot/ardupilot #15732: Copter: FS_THR_DISABLED does not set failsafe.radio

hamzzi, 2020-11-05:
## Bug report

**Issue summary**

If user set `FS_THR_ENABLE` to 0, vehicle skips checking radio failsafe and dose not covert the radio failsafe flag.
The problem is if user set `FS_EKF_ACTION` to 2 which is ALT_HOLD, it checks radio failsafe flag. `FS_EKF_ACTION` to 2  changes mode to ALT_HOLD when EKF failsafe is triggered, but radio failsafe is true, changes mode to LAND.

Problem scenario is below
```
Guided -> [RC failsafe] -> Guided -> [EKF failsafe] -> ALT_HOLD
```
In this scenario, even though RC failsafe is triggered, it will ignored by `FS_THR_ENABLE` and keep the before mode.
And then EKF failsafe is triggered, vehicle will change mode to ALT_HOLD not to LAND.

I think the radio failsafe flag should be changed without changing mode even if the user sets `FS_THR_ENABLE` to 0.

**Issue details**

https://github.com/ArduPilot/ardupilot/blob/5f8b4968ec6eca40d69f441c54bf2a95ea575ebd/ArduCopter/radio.cpp#L131-L152

L134 checks the `g.failsafe_throttle` and return if it is `FS_THR_DISABLED`.
But failsafe flag is set in line L151.

https://github.com/ArduPilot/ardupilot/blob/5f8b4968ec6eca40d69f441c54bf2a95ea575ebd/ArduCopter/ekf_check.cpp#L166-L170

EKF failsafe checks `failsafe.radio in case` of `FS_EKF_ACTION_ALTHOLD`.
If radio failsafe occurred with `FS_THR_ENABLE 0`, it can pass the condition and I think it is not the result we want.

I got the answer ALT_HOLD can be danger if we lost radio transmitter in issue https://github.com/ArduPilot/ardupilot/issues/15346

**Version**
master(f18585c58429efe6b8d5334d8c999864c2f424f7)

**Platform**
[x] Copter

**Airframe type**
Quad Copter

**Hardware type**
SITL

**Logs**
[log.tar.gz](https://github.com/ArduPilot/ardupilot/files/5492042/log.tar.gz)

rmackay9, 2020-11-16:
As mentioned on the PR  I don't really understand the problem here.  The vehicle should not switch to AltHold if an RC failsafe has triggered.  If an RC failsafe has not triggered then the vehicle should be able to switch to AltHold.

hamzzi, 2020-11-16:
As you know, EFK failsafe does not convert to alt hold on RC failsafe. The problem is that if user set FS_THR_ENABLE as 0, rc failsafe does not triggered. So EKF failsafe able to switch to Althold even though throttle power is very low. If rc transmitter has a problem, user cannot control vehicle on althold. I think it is a problem.

rmackay9, 2020-11-16:
@hamzzi why would the user disable the RC failsafe?

hamzzi, 2020-11-16:
@rmackay9  I think some user want just disable RC failsafe action not disable all RC failsafe function.
On the doc, it is clearly stated that FS_THR_ENABLE 0 means disable the radio failsafe entirely. But some user like me think that it is just disable RC failsafe action.
GCS failsafe is same as RC failsafe but `failsafe.gcs` doesn't have much connection with other logic, so it is pretty similar to disable action only, not like RC failsafe.

What I want to say is if user set `FS_EKF_ACTION 2` with `FS_THR_ENABLE 0` have a potential danger.
I think setting proper failsafe option is important, but provide all safe combinations of failsafe options is also important.

Thanks for your interest in this issue.

rmackay9, 2020-11-16:
@hamzzi,

But surely if you're in a manual mode you don't want the vehicle to stay in that mode after an RC failsafe do you?  If you don't want the RC failsafe to trigger during Auto or Guided modes then we have FS_OPTION bits to handle this.

Maybe you hadn't seen these FS_OPTIONS bits?  Given this I still wonder why a user would want to disable the RC failsafe.

hamzzi, 2020-11-17:
@rmackay9 basically you are right. I am trying to say that experts or familiar users will set it up well, but it can be a problem for beginners.
Anyway if you think it is not a problem I will close this issue.
Thank you again for spending your time on this issue.

closed 2020-11-17 by hamzzi

rmackay9, 2020-11-17:
@hamzzi,

Thanks for the report and in-depth discussion!