ArduPilot/ardupilot #15739: Pixhawk 4 Fmu B/E Led Assignment

K1rdro, 2020-11-06:
## Bug report
Original Issue Discussion in Forums:
[https://discuss.ardupilot.org/t/pixhawk-4-fmu-b-e-led-is-blinking-red/38642/2](https://discuss.ardupilot.org/t/pixhawk-4-fmu-b-e-led-is-blinking-red/38642/2 )

**Issue details**

Pixhawk 4 FMU Act (Activity) LED and FMU B/E (Bootloader/Error) LED are swapped in firmware. The red FMU B/E LED flashes during normal activity, and the blue Act LED responds during bootloader & errors.

**Version**
ArduPlane V4.0.7 (f0d8ca5e)

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Fixed-Wing V-Tail

**Hardware type**
Holybro Pixhawk 4

**Logs**
N/A

rmackay9, 2021-08-06:
I've created a PR https://github.com/ArduPilot/ardupilot/pull/18235 that may resolve this issue.

I was wondering if someone could test it for me?  I've put a [copter4.2.0-dev binary here](https://www.dropbox.com/s/ej6c3u9w0z4de6p/copter-420-dev-led-swap-pixhawk4.zip?dl=0) but I could make a plane version if that would be more useful.

closed 2021-08-17 by tridge

ykhedar, 2021-10-18:
Hi,
   Similar issue occurs for PH4-Mini board as well. I uploaded Rover 4.1.0 Firmware and see the B/E Red light on almost all the time even though all functions are working properly.

Thanks.

rhysperry111, 2023-10-21:
I'm having a similar issue with my Pixhawk 6C running the latest stable rover.

nvarona-ied, 2023-12-20:
I'm having a similar issue with my Pixhawk 6C running the latest stable rover 4.4., Is there any way to remove the flashing red LED?

PR #18235 (2021-08-17): AP_HAL_ChibiOS: Pixhawk4 blue and red LEDs swapped
This is a relatively simple fix to swap the red and blue LEDs on the Pixhawk4 and resolve issue https://github.com/ArduPilot/ardupilot/issues/15739.

This was also brought during [Copter-4.1.0 beta testing discussions here](https://discuss.ardupilot.org/t/copter-4-1-0-beta6-released-for-beta-testing/74139/14).

Fixes #15739