ArduPilot/ardupilot #15742: Neopixel LED issues due to using more memory for string buffers in heap and static/global memory space

Notorious811, 2020-11-07:
in the context of free memory space, i have another issue, i have disabled smartRTL and terrain features on omnibusf4pro, after disabling these features i got around 55Kb of free memory from total 192Kb memory on omnibusf4pro board, so i thought to do some experiments with that free memory as i thought memory is a kind of resource which can be good to have if you wanna try some firmware customizations, but in case where i use this free memory space to create string buffers in heap or in static memory space then there seems some issues if i use more than 16-17Kb of free 55Kb memory, using more than 17Kb of 55Kb memory space for string buffers interrupting Neopixel LEDS on servo7/servo8 on omnibusf4pro and they stop working, so there's still more than 35Kbs of free space available and i cannot allocate more string buffers as it causing problems for Neopixel LEDs

_Originally posted by @Notorious811 in https://github.com/ArduPilot/ardupilot/issues/15562#issuecomment-723433730_

IamPete1, 2020-11-07:
Neopixles need DMA capable memory, not all memory is created equal. 

I'm not really sure what your expecting to be done to resolve this issue?

Notorious811, 2020-11-07:
It will be good to have features to know more about memory statistics of different threads/processes at run time, if possible then features or even some documentation to handle(allocate/deallocate/reallocated) different kinds of memories involved(like DMA,CCM etc) at run time or compile time

IamPete1, 2021-07-26:
We now have thread and tasks starts via FTP, https://github.com/ArduPilot/ardupilot/pull/17165 https://github.com/ArduPilot/ardupilot/pull/15236

closed 2021-07-26 by IamPete1