ArduPilot/ardupilot #15752: QuadPlane/Tailsitter: QLoiter speed and angle are completely ignored on roll stick commands

robustini, 2020-11-08:
Tested on "Cat" tailsitter in SITL: in QLoiter mode the commands given with the roll sticks are interpreted in an anomalous way.
QLoiter speeds and angles are not respected, also forcing the max roll angle of the copter.
The problem is related to [this part of the code](https://github.com/ArduPilot/ardupilot/blob/cb034c50a7d84267c8f29ee09e44d0b5f847d06d/libraries/AC_AttitudeControl/AC_PosControl.cpp#L1105), where the tricky bit is rotating the angle limits from body to earth frame.
I enclose a binary log that clearly shows the problem.
Tested with the latest master with PR [15719](https://github.com/ArduPilot/ardupilot/pull/15719), which in any case does not affect the problem described in any way.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tailsitter non-vectored

**Logs**

[2020-11-08 19-31-17.zip](https://github.com/ArduPilot/ardupilot/files/5507053/2020-11-08.19-31-17.zip)

IamPete1, 2020-11-08:
The over speed just seems to be a matter of tune, very poor PSC velocity match.

[image]

The angle is just that is is always using angle max for both roll and pitch rather than using the separate roll and pitch limit param for tail sitters.

robustini, 2020-11-09:
> The angle is just that is is always using angle max for both roll and pitch rather than using the separate roll and pitch limit param for tail sitters.

The situation is even worse, it not only uses the maximum angle Q_ANGLE_MAX, it completely ignores Q_LOIT_ANG_MAX as well.

IamPete1, 2020-11-09:
> it completely ignores Q_LOIT_ANG_MAX as well.

Your Q_LOIT_ANG_MAX is set to 0, so ignoring it is the correct behaviour, it defaults to Q_ANGLE_MAX in that case.

robustini, 2020-11-10:
> 
> 
> > it completely ignores Q_LOIT_ANG_MAX as well.
> 
> Your Q_LOIT_ANG_MAX is set to 0, so ignoring it is the correct behaviour, it defaults to Q_ANGLE_MAX in that case.

Obvious, but if I set Q_LOIT_ANG_MAX to 5 it continues to assume the angle of the Q_ANGLE_MAX, so something big isn't working.

robustini, 2020-11-24:
A video showing the roll issue in QLoiter.
My tx full pitch and full roll controls are visible on both sides.
Pitch behavior is ok, roll behavior is not.
The log I posted above shows the same wrong behavior, maximum angle and loiter speed on the roll are not respected.

https://youtu.be/qR-XPocdnfk

robustini, 2020-12-01:
This issue is fixed with [this commit](https://github.com/ArduPilot/ardupilot/commit/d984ddc1638ded3b408859218a9896aeb0d3446c), thanks to @tridge and Kris!

IamPete1, 2020-12-01:
Although that fix should help it is still not applying the separate tail sitter roll and pitch limits, so I don't think this issue is resolved in full.

robustini, 2020-12-01:
> 
> 
> Although that fix should help it is still not applying the separate tail sitter roll and pitch limits, so I don't think this issue is resolved in full.

True, but at least now the roll is as limited as the pitch.
In case of tailsitter it should be separated if Q_TAILSIT_RLL_MX is defined.

robustini, 2020-12-01:
@IamPete1 you are right, I have tried again and it does not solve completely, it helps but the problem is still there.