ArduPilot/ardupilot #15753: QuadPlane/Tailsitter: wrong behavior in case of RTL inside RTL Radius

robustini, 2020-11-09:
The log shows a rather dangerous behavior in case you activate an RTL with the tailsitter inside the RTL Radius while it is in a copter mode, in this case QLoiter, therefore also in case of battery or rx failsafe.
If the tailsitter is in copter and RTL is activated but is inside RTL Radius theoretically it should position itself on the home point, and once reached start the vertical descent to land, instead look at what happens from the log.
For a moment it tries to lower the nose down and then tries to return to vertical, as if he wants to make a transition where there is no need to, causing a crash when it shouldn't.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Tailsitter non-vectored

**Hardware type**
SITL

**Logs**

[2020-11-09 07-22-38.zip](https://github.com/ArduPilot/ardupilot/files/5508500/2020-11-09.07-22-38.zip)

robustini, 2020-11-09:
Tested with the latest master with PR 15719, which in any case does not affect the problem described in any way, QRTL enabled.

peterbarker, 2020-11-24:
@IamPete1 any thoughts?

robustini, 2020-12-16:
I double checked and the problem is there, it is obviously linked to a wrong logic.
It tries to make the transition from MC to FW and when it reaches the angle where it should be in FW it does the transition again in MC instead.

PR #16466 (2021-02-03): Plane: instantly switch from RTL to QRTL if within radius
Fixes #15753

Current code will go into forward flight for 1 second if switched into RTL within the QRTL radius. Even if we remove the 1 second timeout you still get some forward flight throttle for a loop. The only way to prevent (that I have found) is to check in the enter function. 

[second report the forum](https://discuss.ardupilot.org/t/quadplane-rtl-qrtl-behavior-at-low-altitude-danger/66897?u=iampete)
closed 2021-02-03 by tridge