ArduPilot/ardupilot #15754: UAVCAN ESC Telem resetting

asfishman, 2020-11-09:
## Bug report

**Issue details**

MavProxy shows that certain ESC telem messages are restting

**Version**
Master

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
CubeOrange

Two UAVCAN ESC's are running. Report rates are different on each ESC (ESC1 is around 20 Hz and ESC2 is closer to 10Hz)

[image]

magicrub, 2020-11-09:
What HW ESC is it?

asfishman, 2020-11-09:
Custom

magicrub, 2020-11-09:
Can you confirm that this behavior is different than a known-working UAVCAN ESC?

IamPete1, 2022-09-10:
@asfishman any updates?