ArduPilot/ardupilot #15756: MSP multiple backends fail to initialize

yaapu, 2020-11-09:
## Bug report

Multiple MSP backends fail to initialize, only one is actually enabled.

Example configuration that fails:

serial4_protocol=33
serial5_protocol=32
no MSP OSD

serial4_protocol=33
serial5_protocol=-1
MSP OSD OK

link to issue on [discuss](https://discuss.ardupilot.org/t/msp-protocol-support/57944/412)

**Version**
master

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
all

**Hardware type**
all

yaapu, 2020-11-09:
related PR https://github.com/ArduPilot/ardupilot/pull/15757

Kija, 2020-11-09:
LGTM:
serial4_protocol=33
serial5_protocol=32

Confirmed both the Matek Lidar and DJI OSD are working simultaneously on a Matek405-SE.

tenten8401, 2020-11-10:
@Kija Does that matek lidar + opflow board connect and actually work via MSP on ArduPilot? Didn't know it was supported, might have to order one and use it once the multiple MSP fix is in.. seems like a great cheap option for precision loiter or autolanding :)

Kija, 2020-11-10:
> @Kija Does that matek lidar + opflow board connect and actually work properly via MSP on ArduPilot? Didn't know it was supported, might have to order one and use it once the multiple MSP fix is in.. seems like a great cheap option for precision loiter or autolanding :)

Lidar portion works fine. Lidar range is only 2 meters. Optical flow sensor seems to do something (data is logged) but I still need to calibrate it.

tridge, 2020-12-09:
fixed, thanks @yaapu !

closed 2020-12-09 by tridge