ArduPilot/ardupilot #15759: Correction of barometric pressure using outside temperature

fr-correia, 2020-11-09:
## Feature request

Hi guys. I have an issue with an overheating board. With the increasing temperature, my barometer readings also change. I usually just calibrate it again once the board's temperature is stable, but I know that the pressure being reported is wrong. Ok the altitude is always consistent since it is calculated using the calibration reference, but if I try to calculate pressure altitude, I'm gonna have an error that is the result of the increasing heating of the board. 
So I was thinking if there is anyway to correct this. I was thinking of using an extra temperature sensor that would be outside the plane, or at least stick to the inside of the fuselage, and give me a more accurate environment temperature.
Do you guys know if there is any way to calculate a "correction pressure" that could be subtracted from the barometer values, by finding a relation of its reported temperature or an outside one? 
I would dearly appreciate an explanation or a paper I could read about it!
Thanks! 

Thanks a lot!
**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Submarine

IamPete1, 2020-11-09:
Are you familiar with the correction that is done on copter?

https://ardupilot.org/copter/docs/common-baro-temp-comp.html

Copter only atm, but we could look at adding it to other vehicles.

fr-correia, 2020-11-09:
Well I did try to adapt it to Arduplane, but it didn't work very well. It only worked when armed, and at the same time, also the barometers were being continuously calibrated, so I could not take any conclusions.

WickedShell, 2020-11-09:
If you use a digital airspeed sensor (which is generally recommended anyways) we will use it as the temperature source. This is due to the fact that it's almost always mounted much further away from the core FC and subject to less heating. (It's also possible to basically locate the sensor out in a wing/surface far from any other source of heating).

closed 2020-11-09 by WickedShell

fr-correia, 2020-11-09:
@WickedShell I'm not talking about calibration. I'm focusing on the pressure measured being affected by the surrounding heat of the board.
When it comes to calibration I get what you are saying. The moment the baro is calibrated, the altitude will be corrected because it has a reference value to that specific temperature. When looking at the pressure itself is not the same. 
Imagine this. Before calibration, the temperature of the board is the same as outside temperature, and you have a pressure measured, lets say its **alpha hPa**. 15min later, the board has overheated and on you pressure you now measure **alpha plus beta hPa**. Ofc you calibrate it and now you have **altitude = 0 for a pressure of alpha + beta**.
My question here is: Do you think of anyway I can find the true pressure, not altitude, that my barometer should measure, in this case **alpha hPa**, if I know the outside temperature? It could even be the one measured by the digital airspeed sensor.

WickedShell, 2020-11-09:
My point is that we are using the outside temperature as observed by the airspeed sensor for all of this, which gets you a way to have an offboard temperature sensor. It's still subject to some self heating of course, but it's less signifcant then the onboard sensor.

fr-correia, 2020-11-09:
Yes I know, but the pressure the barometer sensor "feels" is affected by the heated environment, am I right? The surrounding particles of air should be affected by this increase in temperature and affect the measures of pressure itself. Correct me if I'm wrong. I was hoping to analyze not the altitude provided by the baro library, but the sensor's measurement of pressure, which seems to be affected by the temperature.

IamPete1, 2020-11-09:
So your looking to use the barometer as a environmental sensor rather than concerned about the effect the temperature drift has on flight?

fr-correia, 2020-11-09:
Not exactly. I'm trying to use pressure from the sensor to find pressure altitude (altitude relative to the isobaric line of 1013.25mbar and T= 15ºC).
If my pressure varies because of the heat transmitted from the board, it will impact my pressure measurements.

vierfuffzig, 2020-11-09:
using additional external baros for quite some time i can't actually see pressure diverging relevantly from onboard baro, regardless of a ~25°C temperature difference onboard vs external:
[image]