ArduPilot/ardupilot #15813: EKF GSF yaw failure with poor GPS velocity

tridge, 2020-11-15:
I was flying master on a small quad with @gmorph today when we got an emergency yaw reset on EKF3. From the log I believe the yaw was actually OK before the reset, and the GSF yaw was incorrect. The GPS was reporting very poor speed accuracy.
There is some log corruption (poor microSD, and high logging rate). We then flew again with a better SD card, and replay enabled. We managed to reproduce the emergency yaw reset with replay logging, although the size of the yaw reset was much smaller.
Compass match to WMM, using compass yaw as reference:
[image]
emergency yaw reset happened at 15:15:40.799
same graph, but using EKF yaw as reference:
[image]
it is clear that the new yaw after the emergency reset is incorrect.
reported GPS accuracy:
[image]
it is a M6 (3DR) GPS. Note that speed accuracy spikes high after the yaw reset.
velocity innovations that triggered the reset:
[image]
note that the copter was almost stationary at the time, but GPS reported high speed (over 7m/s)
The EKF yaw reset in log 28.BIN caused the copter to fly into a tree (it was in POSHOLD mode). Almost no damage, and it flew again a few minutes later.
Logs are here:
first log (with lots of lost logging, no replay)
http://uav.tridgell.net/GrantQuad/00000028.BIN
second log (with no logging loss, and replay enabled)
http://uav.tridgell.net/GrantQuad/00000002.BIN

andyp1per, 2020-11-15:
I get yaw resets all the time on my miniquad particularly triggered by mag innovations, I tend to do all I can to switch them off as they are never what I want.

gmorph, 2020-11-15:
Of course when this kind of testing occurs this seems to happen.  Always have a long recover pole handy :-) 
[image]

priseborough, 2020-11-15:
@andyp1per Can you explain what you mean by "I get yaw resets all the time on my miniquad particularly triggered by mag innovations" Are you talking about the reset of yaw to the magnetomer, or to the EKF-GSF yaw estinmator which will only happen if loss of navigation due to high velocity innovations has been detected.

andyp1per, 2020-11-15:
@priseborough I have been plagued with emergency yaw resets on my miniquad. I'm not sure of the cause, but carefully calibrating the mag and setting EK3_MAG_CAL = 0 or 2 has made it a lot better.

priseborough, 2020-11-15:
Looking at the second log the GPS has velocity errors larger than indicated by the GPA.SAcc metric which fools the GSF into declaring an acceptable yaw accuracy before it should. However the EKF's GPS accuracy checks declared a GPs glitch event before the reset, eg:

2020-11-15 16:04:58.288 GPS Glitch

[image]

[image]

Adding a patch that prevents the EKF-GSF yaw estimator using GPS data when the GPS inflight accuracy check is failing eliminates the yaw reset on replay, but I need to add the EKF-GSF logging to the replay to verify the effect of effect on the yaw estimator.

tridge, 2020-11-15:
@priseborough I'll fix replay to handle the XKY logging

tridge, 2020-11-15:
> Adding a patch that prevents the EKF-GSF yaw estimator using GPS data when the GPS inflight accuracy check is failing eliminates the yaw reset on replay, but I need to add the EKF-GSF logging to the replay to verify the effect of effect on the yaw estimator.

I don't think that is sufficient. The 28.BIN log declares the GPS glitch one second after the emergency yaw reset. I know there is some log msg loss (lots in fact), but I still think it is clear enough that just checking for declared glitch isn't enough

tridge, 2020-11-15:
the lag in the GPS speed accuracy reporting really is a curse here. Can we afford to delay GSF triggering for another second to give the GPS accuracy reporting time to catch up?

tridge, 2020-11-15:
the convergence of the GSF wasn't great at the time of the GSF reset:
[image]
the emergency yaw reset happened at 15:15:40.799
at the time we still had a pretty wide spread of yaw values. A really good convergence looks more like this:
[image]

tridge, 2020-11-15:
or we could look at the GSF velocity innovations
[image]
we had high velocity innovation on all of the GSF instances at the time it did the yaw reset. The smallest innovation across all the instances was the first one, with 4.5m/s

tridge, 2020-11-15:
maybe something like the average velocity innovation across the GSF instances should be below 2m/s?

priseborough, 2020-11-15:
The yaw estimates to not have to be converged for a valid yaw estimate due to the action fo the GSF and the weightings. Applying a criteria such as the one you suggest would make it too slow to be useful for a post takeoff divergence recovery due to bad compass. I have a solution that uses the glitch indicator to inhibit EKF-GSF use of GPS. It should be tested and ready to submit today.

closed 2020-11-16 by priseborough