ArduPilot/ardupilot #15824: MSP with _OPTIONS = 1 and RSSI_TYPE = 1 causes board freeze

vierfuffzig, 2020-11-16:
Accidentally stumbled upon this following local forum request. see https://github.com/ArduPilot/ardupilot_wiki/issues/3076 for reference.

once you set:
any serial port to PROTOCOL = 33 (MSP)
MSP_OPTIONS = 1 (telemetry bit)
RSSI_TYPE = 1 (analog)

you can reliably trigger a board freeze beyond recovery on subsequent reboot. only resolveable by DFU reboot and fw reflash. 

_Please describe the problem_

**Version**
master from oct 2nd up to current master as of nov 16th 2020

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

**Hardware type**
multiple F4 and F7 type boards (matek 765 wing, matek 405 wing, omnibusF4 etc.)

**Logs**
not applicable

FlyEnjoy, 2020-11-16:
thanks Basti - posting this at the right place, 
I am eager to test any updates 
Rainer

tridge, 2020-11-16:
i've reproduced on a CUAV-X7

tridge, 2020-11-16:
fixed here:
https://github.com/ArduPilot/ardupilot/pull/15827
many thanks for reporting!

PR #15827 (2020-11-16): AP_RSSI: handle init ordering with MSP
this fixes a nullptr dereference on startup when MSP asks for RSSI
data before the RSSI library is initialised

Fixes #15824
commit 5385f258 (2020-11-16): AP_RSSI: handle init ordering with MSP

this fixes a nullptr dereference on startup when MSP asks for RSSI
data before the RSSI library is initialised

Fixes #15824
closed 2020-11-16 by tridge