ArduPilot/ardupilot #15833: Cannot connect to Tekko32 4-in-1 ESCs with Cube Orange

bromeara, 2020-11-17:
## Bug report

**Issue details**

We have run a variety of tests and configurations with the Cube Orange flight controller and the issue at the core of this is that when we have the UAV powered only via battery the ESCs will not connect to the flight controller. However when we power it with both battery power and a micro-USB connected to the flight controller they connect just fine. Please see [this thread](https://discuss.cubepilot.org/t/cube-orange-only-works-with-usb-plugged-in/5074/9) for the rest of our tests and their results. We have also run this same setup with PX4 installed and it works just fine out of the box so it would seem that it is a firmware bug more so than a hardware problem. 

There was also an issue with BLHeli passthrough however this is probably the topic for a different but report(However I still shared the image in the google drive link).

**Version**
4.0.3 and 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad

**Hardware type**
Cube Orange, also tested on Cube Black
4-in-1 Tekko32 Metal ESC
4x KDE 330kv motors

**Logs**
[Video of the issue](https://youtu.be/Ec6GRBTSnWE)
[Log files of a boot-up without micro-USB and images of wiring and parameters](https://drive.google.com/drive/folders/1Nyx8QAlrLAxhPN4aGclekW6CMCGMAah-?usp=sharing)

magicrub, 2020-11-17:
I don't think it has anything to do with USB, it looks more like a timing problem where when powered by USB the autopilot boots up faster/sooner than if it boots up at the same time as everything else.

magicrub, 2020-11-17:
Try adding a bootup delay to see if that helps at all using param BRD_BOOT_DELAY.

bromeara, 2020-11-18:
This seems to work for me, I'll add the solution to our forum posts since its not a bug. Thank you.

closed 2020-11-18 by bromeara