ArduPilot/ardupilot #15844: Rover/Sailboat: add mavlink command to allow changing motor state

rmackay9, 2020-11-19:
An AP sailboat user has asked that we add support for changing the motor state using a mavlink message.  They have a ground station button or a companion computer that makes decisions upon when the motor is required or not.

Currently the motor state can only be set using an RC auxiliary switch.  To see where this code is in AP, search for, "USE_MOTOR_ALWAYS"

peterbarker, 2020-11-19:
```
+      <entry value="218" name="MAV_CMD_DO_AUX_FUNCTION">
+        <description>Execute auxillary function</description>
+        <param index="1" label="AuxillaryFunction">Auxillary Function.</param>
+        <param index="2" label="SwitchPosition" enum="MAV_CMD_DO_AUX_FUNCTION_SWITCH_POSITION">Switch Level.</param>
+        <param index="3">Empty.</param>
+        <param index="4">Empty.</param>
+        <param index="5">Empty.</param>
+        <param index="6">Empty.</param>
+        <param index="7">Empty.</param>
+      </entry>
```

peterbarker, 2020-11-19:
Also actually had code behind it, which also added the option to have a mission item which triggered an arbitrary function:

https://github.com/peterbarker/ardupilot/commits/wip/handle-command-do-aux-function

IamPete1, 2020-11-19:
I have also a PR that can do this as part of a mission.

https://github.com/ArduPilot/mavlink/pull/106
https://github.com/ArduPilot/ardupilot/pull/12774