ArduPilot/ardupilot #15851: Add check for ardupilot/modules being present

maksimdrachov, 2020-11-19:
## Feature request 

**Is your feature request related to a problem? Please describe.**
I forgot to do `--recursive` when cloning the git, this causes errors later on due to the ardupilot/modules folder being empty.

For example, when trying to execute `./waf configure --board=Pixhawk1`, you will get the following error:

[image]

**Describe the solution you'd like**
There should be a check to see if the folders ardupilot/modules/ are empty. If they are, an error message should be thrown to warn the user that the modules haven't been installed yet and propose to execute `git submodule update --init --recursive`.

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Additional context**
I'm new to opensource development, but would like to solve this issue myself with some help. If this proposal is accepted I will start working on a solution.

khancyr, 2020-11-19:
hello,
Thanks for the proposal, I agree that it would be good to have a more explicit error for newcomers !
We already have a check for submodules into https://github.com/ArduPilot/ardupilot/blob/master/Tools/ardupilotwaf/git_submodule.py , you can start from there to extend it and check that the submodules and their submodules are correctly initialized !

maksimdrachov, 2020-11-26:
### First attempt
So my first attempt at solving the problem has some downsides and I think is unfeasible. As suggested I've started looking at git_submodule.py and found that I could expand configure() to check whether the submodule folders are empty, in case they are I issue a command that should solve the problem using subprocess.
[image]
However to check if the folders are empty I had to go into modules/waf/waflib/Configure.py and change the find_file definition to check if a folder is empty in case the filename == None.
[image]
Later I realized that waf is a module and so no part of Ardupilot project and have difficulty committing this file to the repository shown above (I checked .gitignore but found nothing that would prevent from adding the folder to the staging area). You can try using the file here:
[Configure.py.zip](https://github.com/ArduPilot/ardupilot/files/5601587/Configure.py.zip)

### Another solution?
During my search for a solution I have also found the following in ardupilot/waf
[image]
I wonder if something similar could be done to check for the other modules and if this might be an easier (and better) solution.  

If you have some suggestions and/or feedback I would love to hear it!

khancyr, 2020-11-26:
I think you were close on your first solution. Instead of using waf function, just create a new one that suit you needs!

closed 2023-06-04 by maksimdrachov