ArduPilot/ardupilot #15861: No libcsfml-2.4 on upto date debian streatch

syonfox, 2020-11-21:
#15243  Bug report

**Issue details**
Issue install the tools on debian stretch with

 `Tools/environment_install/install-prereqs-ubuntu.sh -y`

**Version**
https://github.com/ArduPilot/ardupilot/commit/fe81387433eae3c3fb356d611bee7d27af2a7e68

**Platform**
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**error**
```E: Unable to locate package libcsfml-audio2.4
E: Couldn't find any package by glob 'libcsfml-audio2.4'
E: Unable to locate package libcsfml-graphics2.4
E: Couldn't find any package by glob 'libcsfml-graphics2.4'
E: Unable to locate package libcsfml-network2.4
E: Couldn't find any package by glob 'libcsfml-network2.4'
E: Unable to locate package libcsfml-system2.4
E: Couldn't find any package by glob 'libcsfml-system2.4'
E: Unable to locate package libcsfml-window2.4
E: Couldn't find any package by glob 'libcsfml-window2.4'
E: Unable to locate package pkg-config-arm-linux-gnueabihf
```
I then ran into missing 
`pkg-config-arm-linux-gnueabi`

I was still running into errors with the libgfortran3 due to  old python so had to change python and pip to version 3

I got around all of those  by adding the following to the config

```
...
    SITLFML_VERSION="2.5"
    SITLCFML_VERSION="2.5"
elif [ ${RELEASE_CODENAME} == 'stretch' ]; then
    SITLFML_VERSION="2.5"
    SITLCFML_VERSION="2.5"
    PYTHON_V="python3"
    PIP="pip3"
elif [ ${RELEASE_CODENAME} == 'focal' ]; then
    SITLFML_VERSION="2.5"
    SITLCFML_VERSION="2.5"

...

if [ ${RELEASE_CODENAME} == 'stretch' ]; then
        ARM_LINUX_PKGS="g++-arm-linux-gnueabihf pkg-config"
else
        ARM_LINUX_PKGS="g++-arm-linux-gnueabihf pkg-config-arm-linux-g>
fi
...

````
Here is the full file but the fix 
[install-prereqs-ubuntu.txt](https://github.com/ArduPilot/ardupilot/files/5576735/install-prereqs-ubuntu.txt)

I supose you guys arent suporting debian stretch directly but ket me know if i should make a pull though is just put my fix here for the next guy

syonfox, 2020-11-21:
I also ran into an ieesue with running `sim_vehicle.py -w`
This fixed it its becuse we need to have pythong scripts executed by python3
```
update-alternatives --install /usr/bin/python python /usr/bin/python3.5 2
```
https://linuxconfig.org/how-to-change-default-python-version-on-debian-9-stretch-linux

khancyr, 2020-11-21:
Hello,

Thanks for the report. Inam already working on it as we got the same issue on the latest ubuntu. The fix will come soon

WickedShell, 2021-04-13:
@khancyr any update if we can close this as resolved by now?

peterbarker, 2024-07-31:
I think we have to close this as too-old now, sorry.  Pretty sure Stretch is EOL now.

Thankfully Py2 vs Py3 issues are mostly behind us....

closed 2024-07-31 by peterbarker