ArduPilot/ardupilot #15864: Occasionally EKF3 estimations going haywire on ground

Notorious811, 2020-11-21:
## Bug report

**Issue details**
I was just checking GPS RTK mode status in open space without flying with multiple quadcopters. When checking the logs, i found something going really wrong with EKF3 estimations after some time, please have a look of POS log, position and altitude estimations almost going haywire after some time since boot up. But these kind of EKF3 inaccuracies does not arise everytime, it happens very rarely, but in this case it has been observed in 2 quadcopters at same time. I have never seen these kind of POS log when i fly these quadcopters.

From the outside it seems like some trivial issue but I have been through some series of events where i am suspecting some serious underlying problem here, i don't know what causing EKF3 estimations going haywire like this, it can also possibly due to some faulty harware/component too but i am seeing this problem on multiple quadcopters. I can clearly see that EKF3 innovations are not happy, but i don't know what will happen if this kind of behaviour happen in air cause there is no warning or failsafe event happened during the time when drones are on the ground.

But on the other hand, i have never seen these kind of EKF3 estmations inaccuracies when i fly multiple quadcopter except in one case which is mentioned in detail here https://discuss.ardupilot.org/t/continuous-uncommanded-climb-suspecting-ekf-issues/63137/48, before this happened, on this particular quadcopter i was facing much more serious EKF3 inaccuracies which is mentioned here https://discuss.ardupilot.org/t/too-large-deviations-between-raw-gps-ekf3-pos-data/62980, i don't know my current issue is having some relationship with it or not

**Version**
ArduCopter v4.0.4

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad-X

**Hardware type**
Omnibusf4pro (Omnibus F4 SD)

**Logs**
Drone1 - https://drive.google.com/file/d/11jMbVzSToNkCCm9ouclIBS9HmSVdUE2_/view?usp=sharing
Drone2 -  https://drive.google.com/file/d/19XQPZ88o2O4_vp9zI9HkiepHP8hhBEB7/view?usp=sharing

IamPete1, 2020-11-21:
Can you try on master, there was a issue that has been fixed if the RTK GPS was reporting it was more accurate that it actually was. Although I cant find the PR now.

Notorious811, 2020-11-21:
@IamPete1 yeah, RTK GPS position and altitude estimations are quite accurate in RTK float mode when EKF3 estimations going crazy, I gonna checkout the EKF PRs in master and try to backport it on arducopter v4.0.4

peterbarker, 2020-11-22:
@Notorious811 the fixes that @IamPete1 was referring to are in Copter-4.0.5

```
commit a40290fb0fb8b72d64435adbe1dba501f9b053a9
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Thu Aug 20 11:31:56 2020 +1000

    AP_NavEKF3: apply min GPS accuracy at measurement point
    
    this fixes an issue a RTK GPS gives 1cm horizontal and vertical
    accuracy and that causes the variances to get too small
```

There are other important bugfixes in that release.  I suggest you move from 4.0.4 to 4.0.5

Notorious811, 2020-11-23:
found the PR #15109, although i had to dug a bit to find it, by just looking at commits and title does not indicate well what really was the problem, conversations help a lot in understanding the issue better, generally in master some commits shows respective PR but some don't, would be ideal if every commit can show their respective PR

peterbarker, 2020-11-23:
On Sun, 22 Nov 2020, Notorious7 wrote:

> found the PR #15109, although i had to dug a bit to find it, by just looking
> at commits and title does not indicate well what really was the problem,
> conversations help a lot in understanding the [...]

github interface allows you to go from commit back to PR.

https://github.com/ArduPilot/ardupilot/commit/35c2d13e2fbf6b1df367c0d247227f29e8af2dc8

Look for      "master (#15834)"

Notorious811, 2020-11-23:
@peterbarker yeah, I was talking about this only, but not all commits shows their respective PR like here https://github.com/ArduPilot/ardupilot/commit/e1a033b29613c5ce4b40354f555316cf74e56e82 or in stable release commits like here https://github.com/ArduPilot/ardupilot/commit/a40290fb0fb8b72d64435adbe1dba501f9b053a9

Notorious811, 2021-03-24:
#16842 solved this issue

closed 2021-03-24 by Notorious811

LeiOuYang, 2021-06-11:
@Notorious811
I also encountered a highly abnormal problem. Does this modification support 4.0.x firmware? At present, there is a stable version of 4.0.x, and 4.1 and above are all tested versions. There may be some unknown problems.
I imagine this known serious bug should be implemented in a stable version.

Notorious811, 2021-06-11:
> @Notorious811
> I also encountered a highly abnormal problem. Does this modification support 4.0.x firmware? At present, there is a stable version of 4.0.x, and 4.1 and above are all tested versions. [...]

this patch has been merged in master so you should try master, no stable releases (4.0.x) of Copter have this patch merged/backported as of yet, you can also check the recent beta release if this patch has been included or not

rmackay9, 2021-06-11:
PR 168442 is included in the Copter-4.1.0-beta1 (and higher)

LeiOuYang, 2021-06-15:
@rmackay9 @Notorious811
Okay! Thanks!