ArduPilot/ardupilot #15876: used cygwin to generate firmware for v5 + using the option ./waf configure --board fmuv5 and ./waf plane. Now when the generated firmware .apj file is uploaded to v5+, in the PFD <<NO GPS>> error is always present.

Gokulk95, 2020-11-23:
**IF YOU DON'T REMOVE THESE FOUR LINES, THEN YOUR REPORT WILL BE CLOSED AUTOMATICALLY**
Questions and user problems should be directed at the forum (http://discuss.ardupilot.org)
_**Please be very sure you have found a bug when opening this issue**_
If there was a previous discussion in the forum, link to it

## Bug report

**Issue details**

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

magicrub, 2020-11-23:
@Gokulk95 Please don't write a paragraph in the subject area. I've re-written it for you.

magicrub, 2020-11-23:
@Gokulk95 You're lacking lots of details that make this not worth looking into yet. 
1) you have custom firmware. Its easy for us to say "you broke it" without sharing your branch.
2) Please be specific on the autopilot hardware. I don't know what "V5+" is.
3) What is your GPS hardware? How is it connected? Is it even enabled in the params?

Gokulk95, 2020-11-24:
@magicrub  
please find the picture of the setup used

Gokulk95, 2020-11-24:
Thank you, it is the first time creating an issue.

v5+  board is commonly known as CUAV V5+. I am attaching the snapshot of the setup. there is no branch because there is no update made. It is just to check if with cygwin i will be able to generate a firmware. So the following steps are used:
git clone https://github.com/ardupilot/ardupilot.git
cd ardupilot
git submodule update --init --recursive
./waf configure --board CUAVv5
./waf plane

[image]

amilcarlucas, 2020-11-26:
@Gokulk95 please change the tittle to "ArduPlane master on CUAV V5+ reports <<NO GPS>> with NeoV2"

amilcarlucas, 2020-11-26:
@tridge can you give me permissions to change the github issue title

IamPete1, 2021-08-16:
This sounds like a configuration issue. @Gokulk95 were you able to resolve?