ArduPilot/ardupilot #15889: run Tools/environment_install/install-prereqs-mac.sh on apple m1 macbook

liang-tang, 2020-11-24:
install environment error
## Bug report

tangliang@tangliangdeAir environment_install % ./install-prereqs-mac.sh 
Checking homebrew...
Homebrew installed
xcode-select: error: command line tools are already installed, use "Software Update" to install updates
Updating Homebrew...
==> Tapping ardupilot/px4
Cloning into '/usr/local/Homebrew/Library/Taps/ardupilot/homebrew-px4'...
remote: Enumerating objects: 242, done.
remote: Total 242 (delta 0), reused 0 (delta 0), pack-reused 242
Receiving objects: 100% (242/242), 305.18 KiB | 612.00 KiB/s, done.
Resolving deltas: 100% (137/137), done.
Error: Invalid formula: /usr/local/Homebrew/Library/Taps/ardupilot/homebrew-px4/kconfig-frontends.rb
kconfig-frontends: Calling a Formula#patches definition is disabled! Use 'patch do' block calls instead.
Please report this issue to the ardupilot/px4 tap (not Homebrew/brew or Homebrew/core), or even better, submit a PR to fix it:
  /usr/local/Homebrew/Library/Taps/ardupilot/homebrew-px4/kconfig-frontends.rb:23

Error: Cannot tap ardupilot/px4: invalid syntax in tap!
Already up-to-date.
Updating Homebrew...
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
Error: No available formula or cask with the name "genromfs".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
Updating Homebrew...
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
Error: No available formula or cask with the name "gcc-arm-none-eabi".
==> Searching for a previously deleted formula (in the last month)...
Error: No previously deleted formula found.
==> Searching taps on GitHub...
Error: No formulae found in taps.
Updating Homebrew...
Error: Cannot install in Homebrew on ARM processor in Intel default prefix (/usr/local)!
Please create a new installation in /opt/homebrew using one of the
"Alternative Installs" from:
  https://docs.brew.sh/Installation
You can migrate your previously installed formula list with:
  brew bundle dump
Checking pip...
pip installed
./install-prereqs-mac.sh: line 29: pip2: command not found
./install-prereqs-mac.sh: line 31: realpath: command not found
usage: dirname path
./install-prereqs-mac.sh: line 32: realpath: command not found
Add /Tools/autotest to your PATH [N/y]?nSkipping adding /Tools/autotest to PATH.
finished
tangliang@tangliangdeAir environment_install % 

_Please describe the problem_

**Version**
_What version was the issue encountered with_

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
_What autopilot hardware was used? (Pixhawk, Cube, Pixracer, Navio2, etc)_

**Logs**
_Please provide a link to any relevant logs that show the issue_

seksitha, 2020-11-29:
Hi, I likely got the same error when trying to setup build environment for mac. 
`Invalid formula: /usr/local/Homebrew/Library/Taps/ardupilot/homebrew-px4/kconfig-frontends.rb`

Magnorio, 2020-12-29:
Hi, I got the same problem on my side!

neom89, 2021-01-14:
Hi, I am getting the same issue. Did you find a solution to the issue?

andyp1per, 2021-01-14:
My suggestion is to use the Rosetta shell - I was able to overcome all my homebrew issues that way

neom89, 2021-01-14:
thanks. I will try that.

neom89, 2021-01-14:
I get the following error on trying to install rosetta
`Installing Rosetta 2 on this system is not supported`

Is there any other way to get around the homebrew issue?

andyp1per, 2021-01-14:
Rosetta is pre-installed:

https://osxdaily.com/2020/11/18/how-run-homebrew-x86-terminal-apple-silicon-mac/

neom89, 2021-01-14:
I tried the process given in the link but there is no box for “Open using Rosetta” in the Get Info window of my terminal on mac.

neom89, 2021-01-15:
@gmorph Can you please give some ideas on how to get around this build error on MacOs?

seksitha, 2021-01-15:
@neom89 You can setup build environment with docker or vagrant look at the docs

IamPete1, 2021-08-16:
It sounds like this has been resolved?