ArduPilot/ardupilot #15899: Airspeed Measurement delay changed from EKF2 to EKF3

fr-correia, 2020-11-25:
## Bug report

**Issue details**

Hi there guys!
Ok so I have recently put my interest on how the EKF works, it is a big and complex estimator and quite honestly, it is very satisfying to see how things are so well packed in the code!
Lately I put my focus on one thing: sensor fusion.
I get the picture that the moment you take a measurement from a sensor, that same measurement is subjected to a delay from its specific sensor. The one thing I don't get it is, how do we find this delay? Imagine I would like to install a different sensor, like 10x better than the ones I'm currently using. I would probably have to change this delay, since it is a different hardware. 
In the current EKF, the barometer delay is accepted as an external parameter "EK2_HGT_DELAY", but when it comes to airspeed or magnetometer fusion, these values are protected and fixed to 240ms and 60ms, respectively.
Also, I noticed that in the EKF3 the value of the airspeed delay is changed to 100ms. 

I'm not sure if this is a bug, to account for a different value in the airspeed delay in the EKF3, or if there is any logic to find this value, I'm not finding... Appreciate all information you can give me ;) Thanks!

**Version**
4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[x] Plane
[  ] Rover
[  ] Submarine

rmackay9, 2020-11-25:
This is a question so it's probably best to raise this in discuss or in discord chat. In particular there is an EKF room.

In general we are trying to move the delays to be held in a parameter in the sensor library itself.  So for example VISO_DELAY_MS has the delay for the external navigation system (i.e. vicon, t265, etc).  Historically the delays were held in parameters in the EKF library (i.e EK2_HGT_DELAY).

For the airspeed sensor, as you say, it appears to be hardcoded to 100ms and mag is hardcoded to 60ms.  It's possible that these constants could be moved to parameters but we would probably need some testing to show that these are not accurate values and/or are different between sensors.

closed 2020-11-25 by rmackay9

fr-correia, 2020-11-26:
Ok, but how do we find these best values for the delays? Do they come described in the sensor datasheet or something? or do we tuned this with a trial and error method?

rmackay9, 2020-11-26:
@fr-correia,

I don't know how to figure out the delay for an airspeed sensor.  For other sensors like visual odometry (aka external nav) we move the vehicle in a sin-wave (i.e. up and down) and then compare the accelerometer values with the output from the sensor.  We assume the accelerometer has no delay and so by comparing the timing of the peaks from the two sensors we can see the lag.

fr-correia, 2020-11-27:
@rmackay9 ,

That's a smart way to do it. I was going crazy around datasheets trying to figure out if I was missing something, so thanks a lot!