ArduPilot/ardupilot #15908: CAN GPS instance assignment

tridge, 2020-11-26:
If you have two CAN GPS modules then at the moment there is no way to guarantee which will be the first GPS and which will be the second. This is especially problematic if the GPS POS parameters are not zero.
We need a way to specify which GPS is which.
Here are a few possibilities:
 - we could add GPS1_DEVID and GPS2_DEVID, like we do for other sensors, then have a GPS_PRI1_ID and GPS_PRI2_ID, similar to compass
 - we could use GPS1_NODE and GPS2_NODE, so it only aligns on CAN node ID

the first method would also allow ordering of serial devices. The 2nd would only work for CAN

tridge, 2020-11-30:
need to cope with same node ID on two different CAN buses

priseborough, 2020-12-01:
I think we should avoid the implication of 'priority' in the parameter name if adopting the DEVID approach. With GPS, the parameters are assigning a SW instance to a physical device, not necessarily assigning a use priority.

pompecukor, 2020-12-29:
Seems this can be closed now @tridge
https://github.com/ArduPilot/ardupilot/pull/16156

IamPete1, 2021-08-16:
closed by https://github.com/ArduPilot/ardupilot/pull/16156

closed 2021-08-16 by IamPete1