ArduPilot/ardupilot #15915: Watchdog Reset Reporting Page

Hwurzburg, 2020-11-27:
This page is for anyone to report a WatchDog reset (WDG in GCS or DataFlash messages). This allows the Dev team to be alerted, and to investigate as soon as possible.

Please enter the firmware used, vehicle type, and the WDG message. If you have a log, please retain it for 30days in case a Dev would like to examine it.

Note: follow up timing on new issue will depend on many factors, so some may not be addressed immediately.

zhaodelong, 2021-03-23:
I reported a watchdog reset incident at https://discuss.ardupilot.org/t/arduplane-4-0-9-froze-in-the-middle-of-flight/69140/2

firmware used: arduplane 4.0.9 stable,
vehicle type: plane
the WDG message: decoded message in the forum link

Could you investigate this? logs and videos in the forum

tridge, 2021-03-25:
This one from Simon Exner 
https://discuss.ardupilot.org/t/watchdog-reset-in-flight-and-ardupilot-does-not-complete-startup-and-hangs/64447/5

giacomo892, 2021-04-14:
Copter 1fe1e532607de07a38ff9eed44dbb3ae9d333a64 
Matek F405-STD
[image]

WDG: T-2 SL0 FL0 FT0 FA0 FTP0 FLR0 FICSR0 MM0 MC0

I think it watchdogged moments later I've switched to auto

Can share logs with EKF_REPLAY enabled.

```
ACRO_BAL_PITCH   1.000000
ACRO_BAL_ROLL    1.000000
ACRO_OPTIONS     0
ACRO_RP_EXPO     0.300000
ACRO_RP_P        4.700000
ACRO_THR_MID     0.000000
ACRO_TRAINER     2
ACRO_YAW_P       3.500000
ACRO_Y_EXPO      0.000000
AHRS_COMP_BETA   0.100000
AHRS_CUSTOM_PIT  0.000000
AHRS_CUSTOM_ROLL 0.000000
AHRS_CUSTOM_YAW  0.000000
AHRS_EKF_TYPE    2
AHRS_GPS_GAIN    1.000000
AHRS_GPS_MINSATS 6
AHRS_GPS_USE     1
AHRS_ORIENTATION 0
AHRS_RP_P        0.200000
AHRS_TRIM_X      0.000000
AHRS_TRIM_Y      0.010000
AHRS_TRIM_Z      0.000000
AHRS_WIND_MAX    0
AHRS_YAW_P       0.200000
ANGLE_MAX        3000
ARMING_ACCTHRESH 0.750000
ARMING_CHECK     1
ARMING_MIS_ITEMS 0
ARMING_RUDDER    2
ATC_ACCEL_P_MAX  110000.000000
ATC_ACCEL_R_MAX  110000.000000
ATC_ACCEL_Y_MAX  19000.000000
ATC_ANGLE_BOOST  1
ATC_ANG_LIM_TC   1.000000
ATC_ANG_PIT_P    18.000000
ATC_ANG_RLL_P    16.200001
ATC_ANG_YAW_P    4.300000
ATC_INPUT_TC     0.150000
ATC_RATE_FF_ENAB 1
ATC_RATE_P_MAX   0.000000
ATC_RATE_R_MAX   0.000000
ATC_RATE_Y_MAX   0.000000
ATC_RAT_PIT_D    0.007500
ATC_RAT_PIT_FF   0.000000
ATC_RAT_PIT_FLTD 20.000000
ATC_RAT_PIT_FLTE 0.000000
ATC_RAT_PIT_FLTT 25.000000
ATC_RAT_PIT_I    0.102661
ATC_RAT_PIT_IMAX 0.500000
ATC_RAT_PIT_P    0.110000
ATC_RAT_PIT_SMAX 0.000000
ATC_RAT_RLL_D    0.006500
ATC_RAT_RLL_FF   0.000000
ATC_RAT_RLL_FLTD 20.000000
ATC_RAT_RLL_FLTE 0.000000
ATC_RAT_RLL_FLTT 25.000000
ATC_RAT_RLL_I    0.079862
ATC_RAT_RLL_IMAX 0.500000
ATC_RAT_RLL_P    0.080000
[... 934 lines cut ...]
TUNE_MAX         0.000000
TUNE_MIN         0.000000
VTX_ENABLE       0
WPNAV_ACCEL      300.000000
WPNAV_ACCEL_Z    150.000000
WPNAV_JERK       1.000000
WPNAV_RADIUS     150.000000
WPNAV_RFND_USE   0
WPNAV_SPEED      10000.000000
WPNAV_SPEED_DN   150.000000
WPNAV_SPEED_UP   250.000000
WP_NAVALT_MIN    0.000000
WP_YAW_BEHAVIOR  1
```

```
QGC WPL 110
0	0	0	16	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	0.000000	1
1	0	3	22	15.000000	0.000000	0.000000	0.000000	0.000000	0.000000	10.000000	1
2	0	3	18	2.000000	0.000000	15.000000	1.000000	44.SCRAMBLED	SCRAMBLED.297341	10.000000	1
3	0	3	16	0.000000	0.000000	0.000000	0.000000	44.SCRAMBLED	SCRAMBLED.297862	13.000000	1
4	0	3	16	0.000000	0.000000	0.000000	0.000000	44.SCRAMBLED	SCRAMBLED.297060	12.000000	1
5	0	0	178	1.000000	14.000000	-1.000000	0.000000	0.000000	0.000000	0.000000	1

```

Redshifft, 2021-05-02:
[image]

WD issues from Initialization on 4.1beta, Copter fell out of the sky on board reset after 5 min flight.

WD errors are obviously detected ok so is there any way to start making the Quad "YAW KICK" every 2 seconds or something to get the Pilots attention?

andyp1per, 2021-05-02:
@Redshifft what flight controller is this?

Redshifft, 2021-05-02:
Little custom Nano F405 own design. First time it's done this in 60 test flight on 4.1 beta - Fault could be anything from a connection splash to HW/SW 
Like the best Air crashes, Poxy goggles were beeping and I was distracted checking them but nothing obviously wrong while flying. It just fell from  a Hover when the board reset. Logs are incomplete but MP shows the flightpath.
No DMA or over load issues

andyp1per, 2021-05-02:
I'm just curious because its the infamous "all zeros" watchdog. Still not got to the bottom of what causes this. H7 flash corruption can, but not on an F405.

Redshifft, 2021-05-02:
I reported it for the reason that it may be possible to add something like I suggested above if detected mid flight.
This is only the second time I've had issue with a F405 board in hundreds of flights. The other one was again at initialization  on a Plane where it set level at 90deg out.
I'm looking into why my arming checks did not pick this up at arm seems a few of the checks were disabled, which is a mystery.

Redshifft, 2021-05-03:
@andyp1per  I got this wrong, somehow after 5min into flight all this watchdog stuff kicked off, not at bootup. I was quite happy until revisited the evidence. 
After the event I found the SDCard had all this rubbish on it (windows could not see it)
I'll leave it there and we will see what other reports come in.
[image]

Notorious811, 2021-05-29:
@rmackay9 don't think it is related to the latest firmware release, I'm getting this issue on stable arducopter-4.0.4 and 4.0.7 releases as well, just wanted a little help to decode what's the source

bugobliterator, 2021-05-29:
@Notorious811 is the issue consistently repeatable?

Notorious811, 2021-05-29:
> @Notorious811 is the issue consistently repeatable?

yes, it is consistently repeatable with both 4.0 and latest 4.1 arducopter versions of firmware

tridge, 2021-06-01:
@Notorious811 do you have a real omnibusf4pro you can reproduce it on?

peterbarker, 2021-06-01:
@Notorious811 can you provide a log - or just run that `WDG` message through `Tools/scripts/decode_watchdog.py` please?

Notorious811, 2021-06-01:
@tridge this issue is not happening on real omnibusf4pro, it is a custom build FC, close to omnibusf4pro clone just with different sensors, this issue is happening every time after few seconds since powering on the board, it is most probably some design fault from our end but we finding it very difficult to debug, this issue keeps arising even with a minimum hwdef.dat configurations attached below which suggest something we must be doing wrong with setting up components around the MCU.

[hwdef.zip](https://github.com/ArduPilot/ardupilot/files/6578449/hwdef.zip)

Notorious811, 2021-06-01:
@peterbarker i have multiple logs but as mentioned here https://ardupilot.org/copter/docs/common-watchdog.html#determining-that-a-watchdog-reset-occurred, in my case none of the logs have this info within MSG messages about watchdog reset event, still i'm attaching some of the recent logs-
https://drive.google.com/drive/folders/1qOnV9pUQ03r2o0J1QETysFUKRjHdAnVf?usp=sharing

> just run that `WDG` message through `Tools/scripts/decode_watchdog.py` please

i also tried that but i don't have all the info in proper format about  FA, ICSR etc, that's why this is all the data i able to decode using the image
  T              Scheduler Task:                -2: Fast loop
  IE             Internal Error Mask:           0: ?????
  IEC           Internal Error Count:         0: 0
  IEL            Internal Error Line:            0: 0
  MM          MAVLink Message:           0: [None]
  MC           MAVLink Command:         0: [None]
  SL             Semaphore Line:               0: Not waiting on semaphore
  FL             Fault Line:                      128: ?????
  FT             Fault Type:            3: HardFault

estogantsev, 2021-08-22:
I've got a bunch of watchdog messages right after landing, all of them were the same:
WDG: T-1 SL0 FL128 FT3 FAE000ED38 FTP180 FLR805F1F

board: omnibusf4pro
fw: plane 4.1.0 beta5

arm switch was still on and dataflash log was being recorded

tridge, 2021-08-22:
> I've got a bunch of watchdog messages right after landing, all of them were the same:
> WDG: T-1 SL0 FL128 FT3 FAE000ED38 FTP180 FLR805F1F

can you please post the bin logs? There should be 2 logs, one from while it was flying, and a 2nd after the watchdog

estogantsev, 2021-08-22:
> can you please post the bin logs? There should be 2 logs, one from while it was flying, and a 2nd after the watchdog

attached are 2 log files from that event
[logs.zip](https://github.com/ArduPilot/ardupilot/files/7027656/logs.zip)

raj241992, 2022-01-18:
I use ArduPlane v 4.1.5 with pixhawk (fmuv3) and I add a serial external AHRS.
when the serial is set in low baud rate speed (<=115200), the code runs fine.
however when I set the baud rate to 460800 the plane code resets and hangs .
if I disconnect the RX pin only ,every thing runs fine and “internal error 0x800” rises
with this msg:

    WDG:T49 SL0 FL129 FT3 FAD7 50C22FTP181 FLR80C7667

if I connect the RX pin after awhile, the sensor works correctly
is there any description to this behavior ? and how can I solve this problem?
best regard.

pratikiitgn, 2024-03-30:
I have resolved it. Kindly share your query to pkprajapati5959@gmail.com