ArduPilot/ardupilot #15919: Rover: pivot turns broken in master?

rmackay9, 2020-11-28:
It seems that at somepoint between Rover-4.0 and 4.1 we have broken pivot turns.

This can be recreated in SITL by running "rover-skid" with all default parameters and then creating a simple back-and-forth mission.  The results are shown below and we can see the vehicle is doing a regular turn at the corners instead of a pivot turn.

[image]

rishabsingh3003, 2020-11-28:
Was #15677 a consequence of this issue?

rmackay9, 2020-11-29:
@rishabsingh3003, I'm really not sure yet actually.  When I did some testing on the weekend I was expecting pivot turns to not work if OA was enabled but I found they weren't working even with OS disabled.  I hope to look into it more this week..

rmackay9, 2020-12-08:
This is fixed in master now so closing

closed 2020-12-08 by rmackay9