ArduPilot/ardupilot #15941: Plane: stable 4.1 release pending features

tridge, 2020-12-01:
This is an issue to gather key issues for the plane 4.1 release

 - [ ] new quadplane landing system #14398
 - [ ] ship landing code #14888 

Pending quadplane features from Kris:
 - [ ] When the safety switch is disabled, the Tilt servos may oscillate considerably once when connected to the battery. This causes the propeller to strike the wing and damage it. Possible solution: Move BoardConfig.init_safety() in system.cpp to run BoardConfig.init_safety() 2 seconds after init has finished.
 - [ ] In some cases, failsafe will change the flight mode before entering failsafe, which results in the AUTO mode not continuing. Need example log

pompecukor, 2021-04-22:
>     * [ ]  hybrid landing with RTL/QRTL from high alt #10559

This is fixed with this: https://github.com/ArduPilot/ardupilot/pull/16977

Hwurzburg, 2021-04-29:
-[ ] fence in Plane: if action is guided to return point, and no point is entered, the median lat/long extremes of any fence is used...this generates out of bounds returns and looping behavior around zone boundaries.....should just RTL IMHO
-[ ] if you lose GPS, and fall back to DCM, there is no GCS message informing you of this while in normal flight

joshanne, 2021-04-29:
I've submitted a PR to attempt to address the spiralling loiter behaviour you mention in point 1.

Hwurzburg, 2021-05-02:
-[ ] fences in all vehicles: can be ENABLED but non selected in TYPE....probably should be a pre-arm for that and a gcs message if live enabled...

Hwurzburg, 2021-05-19:
-[ ] New bootloader fails in trad heli firmware

IamPete1, 2022-05-01:
User was allowed to arm when they should not have been: https://discuss.ardupilot.org/t/rolled-inverted-during-takeoff-mode/84904

Jaaaky, 2025-06-03:
ArduPlane keeps going up when the next waypoint alt is down if previous alt was too high to be achieved
https://github.com/ArduPilot/ardupilot/issues/30227

ilia-enginer, 2025-07-17:
Hello. I have a problem with using ardupilot sub of any version. After flashing the usb firmware, the usb connection with the mission planner is lost. This bug occurs on all stable versions of sub for speedebeef405v3. Only the latest version of the firmware for developers works properly.