ArduPilot/ardupilot #15962: send_velocity in Guided NoGPs Mode

ahmetozkaraca, 2020-12-03:
Hello everyone ı am developing a tethered quadcopter. I want to use send_ned_velocity in No GPS environment. I can control my copter in GUIDED Mode. But in Guided NoGps mode ı can not send this command. My final aim is to make copter can hover in no Gps environment.

rmackay9, 2020-12-03:
Hi @ahmetozkaraca,

So it is a common misunderstanding that Guided-NoGPS is for Non-GPS estimation/control environments but this is not actually what it is for.  Guided-NoGPS only supports attitude control.

For velocity control the EKF must have a position source.  The list of available methods is here: https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html

Once you have one of the non-GPS position sources then you can use regular Guided mode and send it velocity commands.

Wiki page which mentions this: https://ardupilot.org/copter/docs/ac2_guidedmode.html#guided-nogps

closed 2020-12-03 by rmackay9

ahmetozkaraca, 2020-12-03:
Thanks for your reply Randy,
We have a ground tether device and we want to make the copter hover on that
device. We will use tether angles for positioning system. For example if
tether x direction angle is 10° at 10 meter altitude, we will send velocity
to the x direction 1 meter. And finally we make the copter can hover in No
GPs environment. Do you think is it a good method?