ArduPilot/ardupilot #15967: Copter: EKF3 still initialising pre-arm message when trying to arm with no compass

rmackay9, 2020-12-03:
It should be possible to fly in Stabilize, Acro and AltHold modes without a compass or GPS but it seems that EKF3 loses it's attitude estimate fairly regularly making it difficult to arm and fly.

The exact PreArm failure message is: **PreArm: AHRS: EKF3 still initialising**

To recreate the issue with Copter-4.1.0-dev set the following parameters:

- AHRS_EKF_TYPE 3
- EK2_ENABLE 0
- EK3_ENABLE 1
- EK3_SRC1_POSXY 0 (None)
- EK3_SRC1_POSZ 1 (Baro)
- EK3_SRC1_VELXY 0 (None)
- EK3_SRC1_VELZ 0 (None)
- EK3_SRC1_YAW 0 (None)
- ARMING_CHECK 598514 (was 1)
- COMPASS_ENABLE   0 (was 1)
- COMPASS_USE1 0
- COMPASS_USE2 0
- COMPASS_USE3 0
- EK3_GPS_CHECK    0 (was 31)
- EK3_MAG_CAL      0 (was 3)

It is sometimes possible to arm but often it is not.  The EKF viewer looks like this:
[image]

The EKF's reported yaw also seems to swing back and forth.

We have this wiki page which explains how to setup AP to fly without a compass but with a GPS: https://ardupilot.org/copter/docs/common-compassless.html

andyp1per, 2020-12-03:
My experience is slightly different to this - the attitude is On *until you try and arm* and only then do the problems start and the errors appear.

Hwurzburg, 2020-12-06:
In plane, using EKF3 now prevents arming a compass-less plane when COMPASS_USEx are all zero.....this is a behaviour change
and will impact 1MB board users...
In addition, Basti has noticed that YAW diverges between DCM and EKF3 on disarm and not when using EKF2 on compassless planes...

rmackay9, 2020-12-08:
As discussed on the dev call, @priseborough is looking at the issue related to EKF3's fusion of dead-reckoned yaw.

Randy, PeterB, Tridge will take on the EKF3 pre-arm check issue.

priseborough, 2020-12-10:
Flying with only an IMU and Baro is not a use case that I have tested before, although I regularly fly with IMU, Baro, GPS and no compass. There is no fundamental reason why the EKF should not be able to handle this given the correct initialisation and shutting down the appropriate parts of the state vector and covariance matrix. I'll have a look next week.

priseborough, 2020-12-13:
To save time on this, can someone that is having this issue reproduce it with LOG_DISARMED=1 and LOG_REPLAY=1 and attach a link to the log. I should be able to solve it quickly using replay with a suitable log. TXS.

andyp1per, 2020-12-13:
@priseborough - here is an attempt. From a flash log so not sure how good it is: https://www.dropbox.com/s/nijoejy9ejdwegx/log2.bin?dl=0

rmackay9, 2020-12-14:
@priseborough I'm having some trouble immediately reproducing the issue on my CubeBlack today for some reason.  I did easily manage to reproduce the issue of the EKF's yaw rotating around freely and there is a log here: https://www.dropbox.com/s/eohvs991c5mb2ng/ekf3-yaw-spin.BIN?dl=0.  I think the issues are related but I'm not sure.

priseborough, 2020-12-14:
@andyp1per Unfortunately you had LOG_DISARMED = 0 so the log is not suitable for replay

priseborough, 2020-12-14:
@rmackay9 I'm able to reproduce the yaw spin behaviour with a close match on replay of your log so will use that to kick of the investigation.

andyp1per, 2020-12-14:
@priseborough - certain I didn't. Since the log is on a flash chip I have to download via mavlink which means setting LOG_DISARMED to 0 at the end otherwise the download never completes.

closed 2020-12-15 by peterbarker