ArduPilot/ardupilot #15977: EKF: add support for upward facing optical flow

rmackay9, 2020-12-04:
We should add support for upward facing optical flow (paired with an upward facing rangefinder) because it could help with non-GPS navigation especially under bridges which is useful for inspection.

The changes required would include:

- adding a FLOW_ORIENT parameter to support upward (we might want to remove the ORIENT_YAW parameter at the same time
- modify the EKF so that it can consume the upward facing flow and also an upward facing lidar

We might need to consider whether we want to try and support both an upward facing and downward facing flow sensor at the same time

ValentineStone, 2022-11-18:
@rmackay9 what is the status of this? I know that upwards facing lidar is already supported.
Also, if you know, would an upwards facing optical flow work with a downwards facing lidar?

Reasoning is: when inspecting inside tunnels the ground surface is usually flat and featureless, whereas the celling will have support beams, pipes, ventilation shafts and alike. So it is better to fly at a know distance from the floor, using a downwards facing lidar, but navigating with optical flow by the features on the celling.

Currently this have been tested to work with the upwards facing T265 camera and a downwards facing lidar using a modified version of the python script found at https://ardupilot.org/copter/docs/common-vio-tracking-camera.html

But this camera is costly and might be an overkill of a simple optical flow sensor such as [MATEK LIDAR SENSOR 3901-L0X](http://www.mateksys.com/?portfolio=3901-l0x) could do the job.

So could you confirm if this kind of setup is possible?

rmackay9, 2022-11-18:
@ValentineStone,

We have a [WIP PR here](https://github.com/ArduPilot/ardupilot/pull/20038) that adds upwards facing optical flow support but I think it has some unresolved issues.  I can't promise when I will get back to this but I'm sure it will happen eventually.

What you describe is possible, AP's surface tracking has a SURFTRAK_MODE parameter that allows you to specify whether it should use the upward facing or downward facing lidar for altitude control.  This is separate from the EKF's position estimation which (if the above linked PR were merged) flow sensor is used.