ArduPilot/ardupilot #15992: fixed-wing control via /mavros/setpoint_raw/attitude  in Guided mode.

zhanghouxin07, 2020-12-06:
Hello there,

I have a fixed-wing drone that uses apm firmware. When I tried to use** /mavros/setpoint_raw/attitude** to control the drone through **mavros**, I found that apm would output an **abnormal rc/out pwm control value**.

That is, the input (/mavros/setpoint_raw/attitude) is **roll=0, pitch=0, yaw=0**, but the **rudder pwm** output of the drone will **increase abnormally by about 220**. For other control functions, this is an abnormal bug.

Hope it can be resolved soon.

Thank you

stable version

**Platform**

[√] Plane

**Airframe type**
Regular fixed wing & VTOL

**Hardware type**
APM

**Logs**
no log

zhanghouxin07, 2020-12-07:
I have solved this problem by myself.

thx.

amilcarlucas, 2020-12-07:
Care to share the solution?

closed 2020-12-07 by amilcarlucas

MatthewFehl365, 2023-07-28:
i know this is a longshot but what did you do to solve the issue? I am getting the same behavior