ArduPilot/ardupilot #16017: Copter/EKF3: pre-arm check failure if GPS_TYPE = 0

rmackay9, 2020-12-09:
Copter's ahrs pre-arm check is mandatory and does not allow users to arm in any mode including modes that do not require a position estimate if:

- GPS_TYPE = 0
- EK3_SRC1_POSXY/VELXY/VELZ = 3 (GPS)

One could argue that this is a good failure because the system has been configured to use GPS but the GPS has been disabled but in practice this is probably annoying for users who just want to quickly disable their GPS and then fly indoors.

It could also lead to configuration errors because users may forget to change EK3_SRC1_VELXY, VELZ back to GPS if we've forced them to reset it to zero at some point.

The simple fix is simply to remove the gps_required checks in [AP_NavEKF_Source.cpp](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF/AP_NavEKF_Source.cpp#L452).

An alternative fix is to pass a boolean into the ahrs pre-arm checks to indicate whether we are in the mode that requires GPS or not.

granewton, 2020-12-16:
Hi Randy

I am using 4.0.5 and cant find the EK3_SRC1_VELXY, VELZ in the EK3 list. 

Is it somewhere else?

Thanks

rmackay9, 2020-12-17:
@granewton,

Yes, that's right, these parameters are new for 4.1.

arshPratap, 2020-12-26:
Sir, I tried working on this problem and after removing the gps required checks ,the build failed in some tests showing the errors MAV-RESULT:FAILED.

rmackay9, 2021-02-04:
This is resolved in master so closing

closed 2021-02-04 by rmackay9