ArduPilot/ardupilot #16020: GPS Loss Leads to Severe Position Drift

obicons, 2020-12-09:
## Bug report

I originally posted this report [here](https://discuss.ardupilot.org/t/gps-loss-leads-to-severe-position-drift/64976). I am reporting because I was able to recreate in stock SITL environment.

 When GPS fails slightly after takeoff, the position of the vehicle drifts severely. 

To recreate, issue these MAV commands using Gazebo Iris:
```
mode guided
arm throttle
takeoff 20
param set SIM_GPS_DISABLE 1.0
```

**Version**
Head, but I have reasons to suspect this problem affects versions going back to Copter-3.6.9. I found this issue on a customized 3.6.9 build originally.

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Iris copter in SITL.

**Hardware type**
SITL

**Logs**
https://drive.google.com/file/d/1wXKSZpJHw5bgJ6F5NCrBrNa43JmBHYSY/view?usp=sharing

peterbarker, 2020-12-10:
We lose position at a rate proportional to t^2 when using EKF and proportional to t when using DCM - if it is any worse than that, please reopen this issue.

closed 2020-12-10 by peterbarker

obicons, 2020-12-10:
I am seeing a loss that averages out to ~9m/sec that looks proportional to t^2 -- but is this the desired behavior? 9m/sec seems high to me. Attached are my lat/long graphs. Thanks for taking the time to look at this!
[image]
[image]

obicons, 2020-12-10:
Also, would it be appropriate to failover to DCM in this situation, or to have the failover option? Once again, thank you @peterbarker for looking!

rmackay9, 2020-12-10:
@obicons,

DCM doesn't provide the high accuracy that most users expect and the fallback itself could dangerous (in my opinion at least) because the DCM is so different from the EKF and is missing some of the safety features and health reporting that the EKF provides.

We have some disagreements in the team about this but this is the decision we've made until now at least.

I think it's best to look at what the real requirement is.  I don't think "DCM degrades less quickly" is actually good enough for most users.  The requirement is not, "I'd like it to be able to fly for X more seconds" the requirement is, "I'd like it to fly without a GPS at all" which leads to other solutions like optical flow, 3D positioning cameras, etc which we've got improved support for in Copter-4.1.

Sorry for the lengthy answer but it's a question that has come up in the dev team before so I thought I'd give a complete answer for the record.

peterbarker, 2020-12-11:
On Thu, 10 Dec 2020, obicons wrote:

> I am seeing a loss that averages out to ~9m/sec that looks proportional to t^2 -- but is this the desired behavior? 9m/sec seems high to me. Attached are

Well, definitely not desired.

We're basically at the mercy of the gyros, simulated in this 
case.  If you fiddle with the simulated gyro drift you can make 
this better or worse - sadly real gyros/accels don't come with those 
parameters...

Why you don't have a truth-source for position, integrating to get it is 
always going to get out-of-whack.

Feel free to dig in and find bugs / better ways of doing this :-)