ArduPilot/ardupilot #16021: Accelerometer Loss During Takeoff Causes Crash

obicons, 2020-12-09:
## Bug report

In SITL, I am able to trigger a crash when losing the accel just after takeoff. After the crash, the copter's motors do not disarm. The barometer and GPS are still producing good values, so we should at least be able to detect that we have lost altitude and disarm faster.

**Version**
Head, but I have also recreated in Copter-3.6.9.

**Platform**
[  ] All
[  ] AntennaTracker
[ * ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
SITL, Gazebo Iris.

**Hardware type**
SITL, Gazebo Iris.

**Logs**
https://drive.google.com/file/d/1VLTBnnK8zE61JsEsgozFEekgUi82cQi6/view?usp=sharing

amilcarlucas, 2020-12-12:
Even with Ekf core affinity on and multiple access?

Vkd4U2NtVkZOVg, 2020-12-16:
I have a similar crash, when the ekf was enabled. there was a serious gyro mismatch between imu0 and imu1.    imu1 was the fault one, and ekf switch to the wrong imu, then crash.

obicons, 2021-02-20:
I'll look into the behavior with EKF core affinity and multiple access and report back.

obicons, 2021-02-20:
In SITL I can confirm something similar occurs with EKF3. The copter never disarms with EKF3. Because of this, I don't have a log to attach. Is there a way to force log completion pre-disarm, or force disarm?