ArduPilot/ardupilot #16049: Quadplane: VTOL_TAKEOFF altitude it is added to the current one in case of mission restart in flight

robustini, 2020-12-14:
If reset a mission in flight and restart AUTO the declared height VTOL_TAKEOFF is no longer relative to the take-off height but to the current one during the flight.
Example, the quadplane is in CRUISE at a height of 20 meters and the VTOL_TAKEOFF in the mission is 30 meters, if you reset and restart the mission the quadplane will perform the MC->FW transition to 50 meters (30 + 20) and not 30 meters from zero of takeoff.
Tested in the SITL with the latest master.

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quadplane

robustini, 2020-12-14:
This log clearly shows the problem.

[2020-12-14 11-39-47.zip](https://github.com/ArduPilot/ardupilot/files/5688250/2020-12-14.11-39-47.zip)

EERomeo, 2020-12-16:
The Vtol does not respect takeoff, like the plane does!
You have a parameter, 'respect takeoff frame' which is disabled by default, you have to enable it in parameters.

robustini, 2020-12-16:
Damn, I missed that, thanks!

closed 2020-12-16 by robustini