ArduPilot/ardupilot #16069: Quadplane: Pitch/Roll inputs have no effect in auto mode after Q_LAND mission command

cheintz, 2020-12-15:
## Bug report

**Issue details**

When auto-landing a quadplane using a Q_LAND waypoint, the elevator and ailerons respond to stick commands (pitch and roll, which correspond roughly to forward and lateral velocity in position controlled modes), but the quad motors do not. The desired pitch and roll do change in response to the stick commands, but these desired values are not achieved. Yaw control does use the quad motors and appears to track the commanded value. 

The quadcopter documentation describes the ability to "nudge" the vehicle in LAND mode [here](https://ardupilot.org/copter/docs/land-mode.html). 

"If the vehicle has GPS lock the landing controller will attempt to control its horizontal position but the pilot can adjust the target horizontal position just as in Loiter mode."

**Version**
4.0.6

**Platform**
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Loong 4+1 VTOL

**Hardware type**
CUAV V5+

**Logs**

[2020-12-10 14-10-05.zip](https://github.com/ArduPilot/ardupilot/files/5698644/2020-12-10.14-10-05.zip)

robustini, 2020-12-16:
That's the Copter wiki, not the Plane.
Many Copter features are not in the quadplane code, maybe this too.

Hwurzburg, 2020-12-22:
Auto mode mission procedures are not flight  modes....executing a MAV_NAV_VTOL_LAND command is not the same as QLAND flight mode....
QLAND flight mode DOES allow you to re-position the vehicle using the multicopter motors and stick movements in all axis...
A NAV_VTOL_LAND is a separate procedure which assumes that you have properly located the land point to avoid obstacles and therefore no pilot intervention should be required....the documentation for VTOL LANDINGs in the Mission Section of the QuadPlane wiki never states that pilot nudging of position is provided. 
feel free to open a request for the option to allow pilot nudging during NAV_VTOL_LAND, but it would be an enhancement.
Closing as this in not a bug...

closed 2020-12-22 by Hwurzburg

cheintz, 2020-12-30:
If nudging isn't allowed in auto modes, the control surfaces and (logged) desired attitude shouldn't respond to pilot stick movements. 

I'll reference this request in an enhancement issue.