ArduPilot/ardupilot #16082: IR-lock problem

WHKnightZ, 2020-12-17:
## Feature request

Hello,
I've bought MarkOne Beacon V3.0 and IR-LOCK Sensor. I tested Pixhawk 4 Mini with firmware ArduCopter v3.6.1. I can see the red LED on the Sensor board when the camera sees the Baecon. In log file of Pixhawk you can see the Heal parameter returns 1 but the Tacq parameter returns always 0. Here I attach the log file and the config parameters file of Pixhawk. Can anyone help me technically to make this works.

Logfile: https://drive.google.com/file/d/1VAQDWQsYbDTLSk9AI__HuwPYG_-4rDEy/view?usp=sharing
Config file: https://drive.google.com/file/d/1HRqxjeFp_pOsI6upn70HER0yQgDgIACf/view?usp=sharing

Thank you.

**Platform**
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

amilcarlucas, 2020-12-17:
You will need ArduCopter 3.6.12 or 4.0.5
Other versions are not really supported. They have bugs that have been fixed in later versions.
How can we fix a potentially already fixed bug?

rmackay9, 2020-12-17:
@WHKnightZ,

I think maybe the vehicle doesn't have a lidar (aka rangefinder)?  As mentioned on [the wiki](https://ardupilot.org/copter/docs/precision-landing-with-irlock.html) the sonar or lidar is required I'm afraid.

rmackay9, 2020-12-17:
This is really a support question so I think I'll go ahead and close this.  In general support should be raised on the forums first and then once it's a confirmed bug or enhancement request we add it to this issues list.  Sorry to be strict but if we're not then our to-do list just turns into another support forum.

Txs for using AP!

closed 2020-12-17 by rmackay9

amilcarlucas, 2020-12-17:
@WHKnightZ please add the answers from Me and Randy to your discord forum post.
And answer the questions we asked you there too.

WHKnightZ, 2020-12-18:
Thanks for all your answers, I will try to install rangefinder and copter 3.6.12