ArduPilot/ardupilot #16087: MAV_CMD_SET_MESSAGE_INTERVAL Does not work right when multiple messages are set to same interval

gezzuzz, 2020-12-17:
## Bug report
**Issue details**
If you set multiple messages (more then 2) to the same interval using MAV_CMD_SET_MESSAGE_INTERVAL  (ex. 200000 (5hz)) 
Results are not as expect. Only 2 of the messages are report at the 5hz rate...the reset never report or report at rates very slow. 
Seem to only allow 2 messages at the same rate.

But if you offset them....  200000 210000 220000 230000 240000 
Then all messages come across like expect.. 

**Version**
4.0

**Platform**
[X] Submarine

**Airframe type**
Custom

**Hardware type**
Pixhawk 1

IamPete1, 2021-08-16:
@peterbarker Maybe you can comment?

peterbarker, 2021-08-30:
I've created an autotest to reproduce this here: https://github.com/ArduPilot/ardupilot/pull/18505

It fails to reproduce.

More information required - any hints, @gezzuzz ?

IamPete1, 2021-09-11:
Seems like this is fixed, we now have a autotest to stop it breaking again. @gezzuzz If you have some steps to re-produce please re-open.

closed 2021-09-11 by IamPete1