ArduPilot/ardupilot #16098: Pitch controller - roll compensation math irregularity

fr-correia, 2020-12-18:
## Bug report

**Issue details**

Hi guys, I found this problem when looking carefully to the pitch controller. So here is the thing: In the link https://ardupilot.org/plane/docs/roll-pitch-controller-tuning.html , you can find a general picture of the pid controllers used to control the several attitude angles. In the case of the pitch we have this:

[image]

where in the inferior part of the image we can see that the roll compensation is something like: Roll compensation = PTCH2SRV_RLL * gravity / (airspeed * tan(bank angle) * sin(bank angle)).

When you search the code, things are a bit different. In AP_PitchController.cpp  the (tan(bank angle) * sin(bank angle))
 term is inverted and the formula gets to be (gravity/airspeed) * tan(bank angle) * sin(bank angle).

I'm reporting this because either the image is wrong or that term on the pitch controller is inverted. Hope I could be of any help.

**Version**
v4.01

**Platform**
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
ALL

**Hardware type**
ALL

khancyr, 2020-12-18:
hello,

The images are 5y old, so surely outdated. But it is worth checking !

rmackay9, 2020-12-18:
Questions should go to discuss and/or discord instead of the issue list so I'm going to close this.  Sorry to be strict but otherwise our to-do list turns into another support forum.

Apologies, I hope someone has an answer though..

closed 2020-12-18 by rmackay9

fr-correia, 2021-01-05:
Hi @rmackay9 . I understand, still I had no success in having an answer to this issue. At the time, I decided to issue a report here as a bug, because it could potentially be major bug, just in case the figures show the real equations to be used.

kniuk, 2021-01-12:
It looks straight forward to me as more up elevator (pitch up) needs to be added in higher bank angles.
Hence _tan(bank angle) * sin(bank angle)_ must be in nominator.  The way it is in the code. Bug in the drawing IMHO.

fr-correia, 2021-01-12:
Thanks @kniuk ! I think I got it

magicrub, 2021-01-12:
Any volunteers to help update the image?