ArduPilot/ardupilot #16103: AP::FS().mkdir() does not seem to work on certain boards after the early boot process

ntamas, 2020-12-19:
## Bug report

**Issue details**

It seems like `AP::FS().mkdir()` does not work at all when invoked after the autopilot has booted (e.g, from MAVFTP), at least on certain boards.

Steps to reproduce:

1. Take a Pixhawk1 or a CubeBlack, install the latest stable ArduCopter firmware on it (4.0.5 at the time of writing).
2. Wait for the board to boot.
3. Try to create a new folder on the SD card with MAVFTP.

Expected result:

The folder should be created.

Observed result:

The folder creation fails with `errno` set to `EIO` (input/output error).

Additional information:

Invoking `AP::FS().mkdir()` in the logger initialization works correctly, but it fails already when calling it right before the `Ready to FLY` message is printed in `system.cpp`. I managed to track things down to a `bouncebuffer_setup_write()` call that happens when the directory entry is created on the FAT filesystem. More precisely, `AP::FS().mkdir()` calls `f_mkdir()`, which eventually calls `dir_clear()`, which creates a temporary buffer of size 32K and then tries to call `disk_write()` with it. `disk_write()` then calls `sdcWrite()`, which calls `sdc_lld_write_aligned()`, which then tries to allocate a bounce buffer. I suspect that the board runs out of free (contiguous) memory blocks because it probably needs _another_ buffer of size 32K as well, and although the `MEMINFO` messages show that I have more than 64K of memory available, chances are that there aren't any continous chunks.

**Version**

Copter 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**

N/A, happens with a bare autopilot unit.

**Hardware type**

Reproduced with Pixhawk1 and Cube Black. Both have an STM32F427; maybe other boards are also affected.

**Logs**

N/A.

ntamas, 2020-12-19:
Some additional debugging reveals that `dir_clear()` in `ff.c` contains two code paths; one allocates a temporary buffer of size at most 32 KB and writes that to the disk to create a clean directory entry, while the other one uses the window buffer (making no extra memory allocation). The execution takes the first path by default since `FF_USE_LFN` is set to 3. Forcing the execution to take the _other_ code path fixes the behaviour of `AP::FS().mkdir()` so it indeed seems like a low-memory issue.

ntamas, 2020-12-19:
On another drone that I have here on my desk (also equipped with a Pixhawk1), `mkdir` via MAVFTP does not work out-of-the-box even with 91872 bytes free after boot. Lowering the temporary buffer size to 8K or even 16K (from 32K) in `dir_clear()` also solves the problem.

IamPete1, 2021-08-16:
possibly related to https://github.com/ArduPilot/ardupilot/issues/14860 and https://github.com/ArduPilot/ardupilot/issues/14644

ntamas, 2021-09-13:
Could also be related to #18351 (and lowering the temporary buffer size in the FatFS driver might be an easier way to make MAVFTP work on low-memory boards than trying to limit the number of reply buffers on MAVFTP's level).

ntamas, 2021-09-25:
Just tested this on a CubeBlack; the problem still exists in ArduCopter-4.1.0-rc2. Patching `ff.c` in ChibiOS still works, but I had to lower the buffer size down to 4K now:

```patch
diff --git a/ext/fatfs/source/ff.c b/ext/fatfs/source/ff.c
index a378cf1ca..16ffe9b40 100644
--- a/ext/fatfs/source/ff.c
+++ b/ext/fatfs/source/ff.c
@@ -542,7 +542,7 @@ static WCHAR LfnBuf[FF_MAX_LFN + 1];                /* LFN working buffer */
 #define FREE_NAMBUF()  ff_memfree(lfn)
 #endif
 #define LEAVE_MKFS(res)        { if (!work) ff_memfree(buf); return res; }
-#define MAX_MALLOC     0x8000  /* Must be >=FF_MAX_SS */
+#define MAX_MALLOC     0x1000  /* Must be >=FF_MAX_SS */

 #else
 #error Wrong setting of FF_USE_LFN
```

IamPete1, 2021-11-16:
Closed by #19267 and #19271

closed 2021-11-16 by IamPete1