ArduPilot/ardupilot #16112: Wrong/missing IMU in HWID page

hagove, 2020-12-22:
## Bug report

I have ha 3DR Pixhawk Mini, flashed with the latest copter firmware, fmuv03 - 4.0.5
BRD_TYPE set to 5 (Pixhawk Mini)

This board has 2 IMU`s,: MPU9250 and ICM20608

The ICM20608 does not show up in the Mission Planner HWID page.
Instead a MPU6000 (Which is not on this board) show up in the list.

https://discuss.ardupilot.org/t/3dr-pixhawk-mini-which-firmware/62750/3?u=oveh

[image]

I have also tried the Pixhawk firmware.
On this firmware, only the MPU9250 shows up in the HWID page..

**Version**
4.0.5 (Have also tested several other version with same result)

**Platform**
Only tested on Copter

**Airframe type**
NA

**Hardware type**
3DR Pixhawk Mini

**Logs**
See attached picture

magicrub, 2020-12-22:
@meee1 I wonder if this is actually a problem with Mission Planner, and not with the firmware?

auturgy, 2020-12-22:
No, that’s in the code. 20608 will identify as a 6000.

Regards,

James

auturgy, 2020-12-22:
https://github.com/ArduPilot/ardupilot/blob/c75848435eeac4ee75089a488e46a73ef8c4a89e/libraries/AP_InertialSensor/AP_InertialSensor_Invensense.cpp#L220

Regards,

James

magicrub, 2020-12-22:
@auturgy  ahh yes, that would do it. Thanks for finding that! Now, what to do about it?

auturgy, 2020-12-22:
From memory there were issues with id’s for the 20608 and it’s better this way. It’s such a critical library that we’d need a very good reason to make a cosmetic but functional change.

hagove, 2020-12-23:
So this only affects the DevType name in Misson Planner?
It will not affect the gyro scaling or IMU performance in any other way?

auturgy, 2020-12-28:
No, everything is working and the driver is doing the right things. Very observant of you to notice by the way!
Do you mind if I close this?

hagove, 2020-12-29:
No, I don't have any problems with you closing it, when I know why, and that it will function correctly anyway. 
But the wrong IMU type in the HWID page, made me a little bit concerned when first noticed.
Thank you for the clarification.

closed 2021-08-16 by IamPete1