ArduPilot/ardupilot #16126: Copter/Rover: move SRTL to separate thread

rmackay9, 2020-12-23:
As discussed on the 23-Dec-2020 EU Dev Call (with @tridge and @andyp1per) we should move SRTL's background processing to a new low priority thread.  The reason to do this is so that the IO thread (which is really a logging thread) could block waiting for logging which could stop SRTL's background thread from running.

SmartRTL's registers the background processing at the end of the init() method ([see code here](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_SmartRTL/AP_SmartRTL.cpp#L126))

Note that we have not heard reports from users of issues with SRTL but that doesn't necessarily mean there aren't issues.

andyp1per, 2020-12-23:
so I think when https://github.com/ArduPilot/ardupilot/pull/16117 goes in this will be largely gone

IamPete1, 2021-08-16:
https://github.com/ArduPilot/ardupilot/pull/16117 has gone in, is this still a problem?

rmackay9, 2021-08-17:
I think we should still probably move SRTL to a new thread although I guess now it is unlikely to make a difference.