ArduPilot/ardupilot #16128: FrSky telemetry missing on Linux

arikrupnik, 2020-12-23:
## Bug report

**Issue details**

No FrSky telemetry appears to emit from the serial port on Beaglebone Blue.

The serial port appears to work correctly. Queries from the FrSky RX arrive, but AP responds only to messages with ID `0x00`.
[image]

Zoom in on AP's response.
[image]

For reference, here's a trace from an F405 board where telemetry s working.
[image]

I’m using a 232 cable from Amazon. The cable is good (tested on F405). I have it plugged into `UART01`. `UART01` and its corresponding port assignment to `SERIAL2` (`-D /dev/ttyO1`) work. If I connect a GPS to `UART01` and set `SERIAL2_PROTOCOL,5`, I get GPS data. When I’m trying to get yaapu telemetry on this port, I set `SERIAL2_PROTOCOL` to 10.

I posted this question here originally: https://discuss.ardupilot.org/t/frsky-telemetry-yaapu-working-on-f405-failing-on-bb-blue/65495

**Version**
Tested with plane 4.0.7, 4.0.6, built locally on the BBB.

**Platform**
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
all

**Hardware type**
Linux on Beaglebone Blue

arikrupnik, 2020-12-25:
According to @yaapu , this may be a limitation of the Linux serial port timing:
https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/1351
If this is the case, I feel we should document this limitation. I'm happy to make a PR against the Wiki. I'm not sure what page this information should be on. It feels like something that's most relevant to Linux users, but I don't know that there is a page for Linux issues as such.

yaapu, 2021-01-11:
@Hwurzburg any hints?