ArduPilot/ardupilot #16137: YAW behaviour sometimes ignores ROI during mission

CarManBelarus, 2020-12-26:
## Bug report

**Issue details**

pixhawk board, flying mission, copter nose position resets sometimes, do not wollow POI. I use 2D gimbal . drone moves nose forward it movement, not POI, after passing some waypoints. I set POI in mission planner manually, it flies normally for some time, and then resets again.

for example, for log#2 :  after WP 32 drone must still look at POI 29, turning slowly to the left, but it turned right to the direction of movement…  

What i wanted to be and what i have in simulation, using tha same waypoints, copter moves to the top of map and looks to the bottom:
[image]

What I have in fact - it looks in the same direction as moves: 
[image]

**Version**
4.0.4, 4.0.5

**Platform**
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
quadcopter

**Hardware type**
Pixhawk

**Logs**
case 1 : [2020-10-30 11-13-37.zip](https://github.com/ArduPilot/ardupilot/files/5742878/2020-10-30.11-13-37.zip)

case 2 : [2020-12-11 10-57-09.zip](https://github.com/ArduPilot/ardupilot/files/5742874/2020-12-11.10-57-09.zip)

rmackay9, 2020-12-28:
This is really a support issue so it should be raised in the [support forums](https://discuss.ardupilot.org/c/arducopter/copter-40/155) rather than here in the issues list.  Sorry to be strict but if we're not then our to-do list turns into another support forum.

I think the issue is the RC4 input has some glitches in it.  It might be good to invest in a better receiver or perhaps increase the RC4_DZ parameter to 40.
[image]

closed 2020-12-28 by rmackay9

CarManBelarus, 2020-12-28:
> This is really a support issue so it should be raised in the [support forums](https://discuss.ardupilot.org/c/arducopter/copter-40/155) rather than here in the issues list. Sorry to be strict but if [...]
Thank you! I will try it. Why I posted here is because the problem appear after update to 4.0.4, never head to such problems with 3.x.x with the same hardware