ArduPilot/ardupilot #16146: 4.0.6.rc1 Quad shoots into air when switched to AltHold

NitinJSanket, 2020-12-27:
## Bug report

**Issue details**

The quadrotor shoots into the air with full throttle when switched from Stabilize to AltHold or Loiter. Using TFMini lidar indoor without GPS on a Holybro Kakute F7.

**Version**
4.0.6.rc1

**Platform**
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
Quad X4.

**Hardware type**
Holybro Kakute F7

**Logs**
https://drive.google.com/file/d/1YiMtv_jK7PZo-0LnQXtjZay-OFNNc5YN/view?usp=sharing

auturgy, 2020-12-27:
This is almost certainly a user configuration issue not a bug, so please raise in the forum - discuss.ardupilot.org

closed 2020-12-27 by auturgy

NitinJSanket, 2020-12-27:
I'm pretty sure it's not a user bug. I am attaching a few more snapshots. The AHR2 altitude looks perfect but the CTUN Alt is garbage.
[image]
[image]