ArduPilot/ardupilot #16148: 4.0.6.rc1 Quad shoots into air when switched to AltHold: Probably an issue with EK3

NitinJSanket, 2020-12-27:
Issue details

The quadrotor shoots into the air with full throttle when switched from Stabilize to AltHold or Loiter. Using TFMini lidar indoor without GPS on a Holybro Kakute F7.

Version
4.0.6.rc1

Platform
[ ] All
[ ] AntennaTracker
[ x ] Copter
[ ] Plane
[ ] Rover
[ ] Submarine

Airframe type
Quad X4.

Hardware type
Holybro Kakute F7

Logs
https://drive.google.com/file/d/1YiMtv_jK7PZo-0LnQXtjZay-OFNNc5YN/view?usp=sharing

It seems like the altitude value is shooting to -150.
[image]

I'm pretty sure it's not a user bug. I am attaching a few more snapshots. The AHR2 altitude looks perfect but the CTUN Alt is garbage.

[image]
[image]

andyp1per, 2020-12-27:
Can you post a .bin file. Most likely explanation is vibration. But best to ask these questions in the forums.

NitinJSanket, 2020-12-27:
Here's the bin file: https://drive.google.com/file/d/1BOaGTTJEUP5UhDzIhoso3NpnNW8PG2ZC/view?usp=sharing
I did do a check for vibrations and they seem to be under 30 all the time.

NitinJSanket, 2020-12-27:
> Can you post a .bin file. Most likely explanation is vibration. But best to ask these questions in the forums.

Here's the bin file: https://drive.google.com/file/d/1BOaGTTJEUP5UhDzIhoso3NpnNW8PG2ZC/view?usp=sharing
I did do a check for vibrations and they seem to be under 30 all the time.

andyp1per, 2020-12-27:
Too much vibration IMO, if you look at IMU.AccZ the variation is huge - that's going to cause the EKF a lot of issues

rmackay9, 2020-12-27:
I agree that the issue is probably vibration although the vibe message isn't terrible and the clip count is low.

The setup is a bit unusual in that the EK3_ALT_SOURCE has been set to 1 (rangefinder) which is nearly always a bad idea.  It's also using EKF3 and optical flow though so perhaps it is an indoor environment.

In any case this is really a support issue so it should be posted in the forums. https://discuss.ardupilot.org/c/arducopter/copter-40/155.  Sorry to be strict but if we're not then our to-do list turns into another support forum.

closed 2020-12-27 by rmackay9