ArduPilot/ardupilot #16162: UAVCAN: Unable to communicate with HERE2 GPS, CubeBlack

jmachuca77, 2020-12-29:
##Bug report

**Issue details**

Here2 GPS is not detected on boot, no CAN communication seems to be going through. If I hot unplug and then replug the GPS, I can get the LEDs to start working but not the GPS. 
Originally the problem was I was trying to flash an early batch Here2 with the CAN boot loader, and then update it to the latest firmware. Using plane 4.0.7 stable and master I was unable to get the SLCAN interface working through mission planner. @meee1 suggested trying early 4.0 release, I loaded copter 4.0 stable and it worked on the first try. Loaded boot loader and then flashed the here2 with the latest firmware. After this I went back to plane 4.0.7, but the GPS was not detected on Boot. Mission planner reporting "NO GPS", and no GPS detected messages on the message tab, and no LEDs flashing on the GPS (NTF_LED_TYPES 231). Without powering down the Cube, I unplugged the GPS and plugged it back in and after the breathing blue lights, the LEDs started flashing normally. Still "no gps" on mission planner and no gps detected messages though. Went to plane master and I have the same exact behavior.
Starting up powered via battery, or USB, same result. 

**Version**
Tested on:
Plane 4.0.7
Master Plane, copter, rover

**Platform**
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Submarine

**Airframe type**
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

**Hardware type**
CubeBlack with Edison carrier, Here2 (HW v2.0 reported, tested with beta firmware, and stable)

jmachuca77, 2020-12-29:
More information. This issue is present on plane firmware 4.0.7 but it does not seem to be there for the latest copter release 4.0.5
The following sequence reproduces the issue for m:
uploader.py arducopter-4.0.5.apj
uploader.py arduplane-4.0.7.apj
mavproxy.py --master /dev/tty.usbmodem14501,115200
param set CAN_P1_DRIVER 1
param set NTF_LED_TYPES 231
param set GPS_TYPE 9
reboot
--- Result no CAN on boot. Unplug and Re-Plug the GPS without powering down the Cube, CAN starts working.

uploader.py arduplane-4.0.6.apj
--- Result CAN works as expected. (No parameters had to be changed because they were retained)

uploader.py arduplane-4.0.7.apj
--- Result CAN works as expected. (there is a bit of a longer delay at boot between when the blue lights turn off to when the LEDs start flashing their normal patterns, about 5 sec, vs 4.0.6 where it is almost immediately after the blue LEDs turn off that the LEDs start flashing in their normal patterns. No parameters had to be changed as they were retained)

bugobliterator, 2021-05-21:
@jmachuca77 I followed the mentioned steps and failed to reproduce the issue. Can you double check if this is still an ongoing issue.

IamPete1, 2021-08-16:
Sounds like this is fixed as @bugobliterator cant reproduce

closed 2021-08-16 by IamPete1